Keychron/qmk_firmware #136: [Bug] Lighning issues with Keychron Q6

Foxlider, 2023-06-29:
### Describe the Bug

Underglow effects with an ISO Knob Q6 are not working properly. 

| Underglow Name | Test |
|--------|--------|
| 01. SOLID_COLOR | OK|
| 02. BREATHING | OK |
| 03. BAND_SPIRAL_VAL | OK |
| 04. CYCLE_ALL | OK |
| 05. CYCLE_LEFT_RIGHT | NOK (Swapped directions) | 
| 06. CYCLE_UP_DOWN | NOK (Swapped directions) |
| 07. RAINBOW_MOVING_CHEVRON | OK |
| 08. CYCLE_ALL | OK |
| 09. CYCLE_OUT_IN_DUAL | Unsure |
| 10. CYCLE_PINWHEEL | OK |
| 11. CYCLE_SPIRAL | OK |
| 12. DUAL_BEACON | OK |
| 13. RAINBOW_BEACON | NOK (Random fixed colors) |
| 14. JELLYBEAN_RAINBOW | NOK (Random fixed colors) |
| 15. PIXEL_RAIN | NOK (Random fixed colors) |
| 16. TYPING_HEATMAP | NOK (Random fixed colors) |
| 17. DIGITAL_RAIN | NOK (Random fixed colors) |
| 18. REACTIVE_SIMPLE | NOK (Random fixed colors) |
| 19. REACTIVE_MULTIWIDE | NOK (Random fixed colors) |
| 20. REACTIVE_MULTINEXUS | NOK (Random fixed colors) |
| 21. SPLASH | NOK (Random fixed colors) |
| 22. SOLID_SPLASH | NOK (Random fixed colors) |

### Keyboard Used

keychron/q6/iso_enmcoder

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q6-qmk-custom-mechanical-keyboard?variant=39990023880793

### Operating System

Windows 11 22H2 22621.1848

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.                                                                                   
Ψ CLI version: 1.1.1                                                                                                         
Ψ QMK home: C:/Users/Keelah/qmk_firmware                                                                                     
Ψ Detected Windows 10 (10.0.22621).                                                                                          
Ψ QMK MSYS version: 1.7.2                                                                                                    
Ψ Git branch: master                                                                                                         
Ψ Repo version: 0.21.3                                                                                                       
Ψ - Latest master: 2023-06-29 08:08:23 +0800 (b5da6d228f) -- sofle/rev1:devdev: Disable console (#21401)                     
Ψ - Latest upstream/master: 2023-06-29 08:08:23 +0800 (b5da6d228f) -- sofle/rev1:devdev: Disable console (#21401)            
Ψ - Latest upstream/develop: None                                                                                            
Ψ - Common ancestor with upstream/master: 2023-06-29 08:08:23 +0800 (b5da6d228f) -- sofle/rev1:devdev: Disable console (#2140
1)                                                                                                                           
Ψ - Common ancestor with upstream/develop: None                                                                              
Ψ All dependencies are installed.                                                                                            
Ψ Found arm-none-eabi-gcc version 10.1.0                                                                                     
Ψ Found avr-gcc version 8.5.0                                                                                                
Ψ Found avrdude version 7.0                                                                                                  
Ψ Found dfu-programmer version 0.7.2                                                                                         
Ψ Found dfu-util version 0.11                                                                                                
Ψ Submodules are up to date.                                                                                                 
Ψ Submodule status:                                                                                                          
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)                                                                   
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)                                                            
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)                                                                 
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)                                                                      
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)                                                                        
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)                                                                      
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)                                                                    
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)                                                                       
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lalalademaxiya1, 2023-07-10:
@Foxlider HI, all the lighting effects of Q6 iso_encoder are working fine on my machine and also correct in VIA. Can you provide relevant videos as materials for analysis? Thank you for your attention to Keychron. I look forward to your feedback. Thank you!

Foxlider, 2023-07-22:
Would that help ? 

[GDrive Video Link](https://drive.google.com/file/d/1-7GYqZLq14vFVca0MctjwtsHngx9kN4r/view?usp=drive_link)

lalalademaxiya1, 2023-07-23:
Sorry, I cannot reproduce your issue on my machine. Could you please take a photo of the version number on the back of the PCBA? It should be located near the middle, slightly towards the top.

lalalademaxiya1, 2023-07-23:
Additionally, could you please take a more detailed instructional video? I would like to better understand the situation you are facing as much as possible. I apologize for any inconvenience I may have caused you.

HannahVernon, 2023-08-22:
I just had the same issue (lighting tab in Via doesn't show the same set of effects as the keyboard itself) for my brand new Q10.  I followed the directions to update the firmware to v1.3 using the QMK toolbox, and that seems to have fixed the issue.  Now, cycling through the possible RGB backlight options exactly matches up with what is shown on the https://usevia.com site.

Foxlider, 2023-08-23:
> Sorry, I cannot reproduce your issue on my machine. Could you please take a photo of the version number on the back of the PCBA? It should be located near the middle, slightly towards the top.

The back of the PCB says `Keychron Q6C V1.2 | ISO | Keychron QMK keyboard | more to come`

> I just had the same issue (lighting tab in Via doesn't show the same set of effects as the keyboard itself) for my brand new Q10. I followed the directions to update the firmware to v1.3 using the [...]

Should I try to update the firmware, then ?