Keychron/qmk_firmware #142: [Bug] 'new-keymap' command not recognized in QMK CLI

taboopython, 2023-07-08:
### Describe the Bug

Hello,

I've been trying to create a new keymap using the QMK CLI, but I'm encountering an issue where the new-keymap command is not recognized. This issue persists whether I use the CLI directly on my system(Linux Ubuntu22.04) or within a Docker container.

Here are the steps I've taken:

Installed QMK CLI using pip3 (pip3 install qmk).
Checked the version of QMK CLI using qmk --version, which returned 1.1.2.
Attempted to create a new keymap using qmk new-keymap -kb keychron/k6_pro -km my_keymap.
The above steps resulted in the following error:

```
error code
usage: qmk [-h] [-V] [-v] [--datetime-fmt DATETIME_FMT] [--log-fmt LOG_FMT]
           [--log-file-fmt LOG_FILE_FMT]
           [--log-file-level {debug,info,warning,error,critical}]
           [--log-file LOG_FILE] [--color] [--no-color] [--unicode] [--no-unicode]
           [--interactive] [--config-file CONFIG_FILE]
           {config,clone,console,env,setup} ...
qmk: error: argument {config,clone,console,env,setup}: invalid choice: 'new-keymap' (choose from 'config', 'clone', 'console', 'env', 'setup')
```

I've also tried updating the QMK CLI using pip3 install --upgrade qmk, but the issue persists.

Any help or guidance on this issue would be greatly appreciated.

Thank you!

### Keyboard Used

Keychron K6 Pro

### Link to product page (if applicable)

https://keychron.jp/products/keychron-k6-pro-qmk-via-%E3%83%AF%E3%82%A4%E3%83%A4%E3%83%AC%E3%82%B9%E3%82%AB%E3%82%B9%E3%82%BF%E3%83%A0%E3%83%A1%E3%82%AB%E3%83%8B%E3%82%AB%E3%83%AB%E3%82%AD%E3%83%BC%E3%83%9C%E3%83%BC%E3%83%89-1

### Operating System

Linux Ubuntu 22.04

### qmk doctor Output

```text
qmk doctor
usage: qmk [-h] [-V] [-v] [--datetime-fmt DATETIME_FMT] [--log-fmt LOG_FMT]
           [--log-file-fmt LOG_FILE_FMT]
           [--log-file-level {debug,info,warning,error,critical}]
           [--log-file LOG_FILE] [--color] [--no-color] [--unicode] [--no-unicode]
           [--interactive] [--config-file CONFIG_FILE]
           {config,clone,console,env,setup} ...
qmk: error: argument {config,clone,console,env,setup}: invalid choice: 'doctor' (choose from 'config', 'clone', 'console', 'env', 'setup')
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lalalademaxiya1, 2023-07-10:
@taboopython Hi, I think you can use the command 'qmk new-keymap -kb keychron/k6_pro/ansi/rgb -km my_keymap' to have another a try. In fact, under the k6_pro, there are further subdivisions such as ansi/rgb, ansi/white, iso/rgb, iso/white, and so on for different keyboard series.

taboopython, 2023-07-10:
Thank you very much. However, I have tried running it and get the same error.

Unfortunately, I am going to give up development on Linux for now.

lalalademaxiya1, 2023-07-11:
Sorry, I have tried the same command and got the different result like the follow picture, maybe you should run 'qmk setup' to check the qmk environment.
[image]