Keychron/qmk_firmware #145: When will the `bluetooth_playground` branch be merged into master and in the main QMK repo?

edianannink, 2023-07-19:
### Issue Description

Hi all, I'm a very happy K2 Pro user. Sometime ago I was searching for the firmware of this keyboard in the [qmk/qmk_firmware](https://github.com/qmk/qmk_firmware) repo. I couldn't find it and saw that this repo existed. The firmware of this keyboard lives in the `bluetooth_playground` branch and it took me some time to find it. It builds without any problems so that's nice. However, I must say that I'm a little bit disappointed that the firmware of the K series (and more specifically the `bluetooth_playground` branch isn't merged yet into master of this repo and finally in the main [qmk/qmk_firmware](https://github.com/qmk/qmk_firmware) repo. Are there any plans to do this? It would really help to get some traction for the K series keyboards in the QMK community.

uzemelteteshj, 2023-07-25:
Try this run qmk msys `qmk setup -b bluetooth_playground Keychron/qmk_firmware`
after `qmk list-keyboards` and you will see k2 pro

svenjacobs, 2023-07-25:
> Try this run qmk msys `qmk setup -b bluetooth_playground Keychron/qmk_firmware`
after `qmk list-keyboards~` and you will see k2 pro

Unfortunately this doesn't answer the OP's question 🙁 I'm an owner of a K3 Pro and would also like to know when the `bluetooth_playground` is merged to `master` and then to QMK upstream to benefit from new QMK functionality and bugfixes?

ilmagico, 2023-07-26:
I just bought a K3 pro (awesome keyboard btw, typing this with it) and, for a keyboard that advertises as one of its selling point "customizable with QMK" I was really surprised to see that the official QMK repo didn't support it. I found the `bluetooth_playground` branch by accident while troubleshooting an issue with via. This really deserves to be in the main QMK repo, especially for a keyboard explicitly sold as "QMK programmable".

16kRAMpack, 2023-07-26:
(Keychron user, not a developer, but I have hacked around on my own fork of the QMK firmware).

I think there's two issues here:

1. Why doesn't bluetooth_playground get merged into this repo's master?
2. Why isn't the upstream repo (qmk/qmk_firmware) more up-to-date?

Regarding issue 1, [QMK tell developers (like Keychron)](https://docs.qmk.fm/#/newbs_git_using_your_master_branch):

> It is highly recommended for QMK development, regardless of what is being done or where, to keep your master branch updated, but never commit to it. Instead, do all your changes in a development branch and issue pull requests from your branches when you’re developing.

So keychron seem to be doing the correct thing here, developing in a separate branch.

Issue 2 seems to me to be with upstream (QMK). It looks to me like Keychron are submitting [pull requests](https://github.com/qmk/qmk_firmware/pulls?q=is:pr+is:open+keychron), and these are being actioned by QMK devs - but not as rapidly as we might want. Take a look at [this actioned pull request for the C2 Pro](https://github.com/qmk/qmk_firmware/pull/20701) - it's taken nearly three months, and includes comments from keychron-dev like "hi, please check this, it has been so long".

My own keyboard, a K8 Pro, is still in bluetooth_playground. It's unfortunate, but QMK is an open source project and is heavily reliant on volunteers. I want a QMK keyboard (I lean heavily on personal hacks I do through QMK), so my choice is either accept the issues that open source development upstream brings, or volunteer with QMK.

tl;dr - I don't think bluetooth_playground will ever (or should ever) be merged into this repo's master. And merges to the QMK repo seem to me to be issues with upstream.

svenjacobs, 2023-07-26:
>  I don't think bluetooth_playground will ever (or should ever) be merged into this repo's master

Why shouldn't `bluetooth_playground` at least be merged into this repo's `master`? I mean these are actual products out there, sold to customers. Having this in `master` should simplify fetching changes from upstream?

ilmagico, 2023-07-26:
@svenjacobs I think @16kRAMpack is right here. Different projects have different processes they follow to get external contributions done, and if QMK would rather you do PR's from different branches, and never commit to master, that should be followed. QMK sets the rules here, and it's good Keychron is following them.

Instead, I think the main effort should go into merging this into QMK's main repo, and I don't know who's to blame here, but if a product is sold as "QMK supported" or whatever, I'd expect it to be from the official repo.

Maybe the only thing that could be done here is update the README to point users to branches of interest for specific keyboard models, until those get merged upstream. This could be done in a dedicated branch (to follow QMK's rule) and that branch be set as the default one.

svenjacobs, 2023-07-27:
> and if QMK would rather you do PR's from different branches, and never commit to master, that should be followed

I misunderstood their recommendation. I thought they meant their `master` not the `master` branch of the fork.

So it would be nice if Keychron could get their `bluetooth_playground` merged into QMK's `master`. I believe this is in the interest of everyone?

edianannink, 2023-07-31:
Thanks everyone for clarifying and discussing the issue. I think we should create a PR to merge the `bluetooth_playground` into QMK's `master` then if I understand correctly?

16kRAMpack, 2023-07-31:
> Thanks everyone for clarifying and discussing the issue. I think we should create a PR to merge the `bluetooth_playground` into QMK's `master` then if I understand correctly?

Not exactly. `bluetooth_playground` is a development branch, so it's always going to be changing, everyytime Keychron create a new BT keyboard. The process looks to me to be that PRs are created for individual keyboards. (Here's the list of open PRs for Keychron boards I posted earlier: [is:pr is:open keychron ](https://github.com/qmk/qmk_firmware/pulls?q=is:pr+keychron+is:closed) - 13 open as I write this comment).

A PR for the [C2 Pro was merged](https://github.com/qmk/qmk_firmware/pull/21581) a few days ago, so PRs _are_ happening. As I mentioned in my previous comment, based on the list of open PRs and that comment from a Keychron dev (_"hi, please check this, it has been so long"_), it looks to me like the issue is simply that upstream (QMK) aren't able to review and approve PRs all that quickly. It makes sense - they're dealing with PRs from more than just Keychron, and are themselves volunteers on an open source project.

Syphdias, 2023-08-02:
Maybe I missed it but the Q1 Pro (ISO layout) was released before the C1/C2 Pro and there is no merge request open or closed for it. If you search for "keychron q1 pro" you will find one PR where [I asked if this was for Q1 Pro](https://github.com/qmk/qmk_firmware/pull/19307#issuecomment-1494930288), and [it wasn't](https://github.com/qmk/qmk_firmware/pull/19307#issuecomment-1495301347).

Maybe it was forgotten or they plan to do a combined merge for multiple keyboard or there is little ISO love or they only want a few open PRs to upstream at a time. As far as I can tell there are only two people working on getting all Keychron keyboards to work and I hugely appreciate them. They are probably under pressure to prioritise the latest releases but it feels bad to be left out here.

Sadly, I currently lack the skill and time to look into it myself but I am looking forward to the day Q1 Pro ISO gets a PR to upstream and gets merged into upstream eventually.

rian-reis, 2023-08-24:
Did anyone try to merge locally master into bluetooth_playground and build a firmware to see if it works?

mutageneral, 2023-09-01:
I am a new owner of a K2 Pro ANSI RGB and new user of QMK. When using the online VIA tool, I saw this message in the macro section:
> Upgrade firmware to use delays

Does the delay feature currently exist in the bluetooth_playground firmware and the keyboard came with an older version, or is the delay feature still missing? People here have mentioned the bluetooth_playground firmware missing newer features.

On the [download page on the Keychron website](https://www.keychron.com/blogs/archived/k2-pro-factory-reset-and-firmware-flash), it says that the version of the firmware for K2 Pro ANSI RGB is "version 1.00 updated December 20, 2022". Would this have the delay features? Or maybe the delay feature was added to this repo, but a new release was not made so compiling from source would work?

Sorry if this is off-topic.

Draxy518, 2023-09-02:
I'm fairly certain that we're never going to get the bluetooth_playground branch merged into master qmk repo and it's all got to do with the licensing of the bluetooth chip SDKs. The comments on this reddit post discuss the issue: https://www.reddit.com/r/olkb/comments/10ckhi6/2023_and_no_mass_market_bluetooth_olkb/

If that is real issue then I think we should work on getting Keychron to merge the changes from the qmk master into their bluetooth_playground branch so we get the new features.

hedgss, 2023-09-30:
I just bought K11 PRO. 
This seems to be an unfair advertisement to mislead buyers.
Buying a keyboard that declares QMK support, the user expects to have access to all QMK features, including the latest updates and bugfixes.
It looks pretty disappointing.

SmetDenis, 2023-09-30:
I have K 15 Pro and the same issue... :(

svenjacobs, 2023-09-30:
> I'm fairly certain that we're never going to get the bluetooth_playground branch merged into the master qmk repo and it's all got to do with the licensing of the bluetooth chip SDKs. 

I don't understand. If the `bluetooth_playground` branch contains code that is not compatible with GPLv2, it shouldn't even be allowed to be there because the project is licensed under GPLv2 according to the `LICENSE` file in the root folder of this project.

Anyway, if the merge to upstream is not possible for some reason, Keychron should at least keep the branch in sync with QMK's main branch.

I also bought my keyboard mainly because of QMK support and was hoping for regular updates to benefit from latest features and fixes.

adophoxia, 2023-09-30:
Seeing as there's some... confusion around this topic, I'd hope to bring some clarification into this.

This was never, not an issue with incompatible licenses. The issue with this whole ordeal is because the code that Keychron used to make BT work for their K Pro and Q Pro boards is too custom to where even in its current state, it's not mergable. 

Both, KC and and the QMK team are still trying to work out a solution towards this. It may be possible when the QMK team merge RIOT OS support into the repo, which is needed for their XAP project (XAP is their variant of VIA, but without relying on a third party for them to have reason to maintain support.) 

So until/if this happens, people who want custom firmware for these kinds of boards will have to rely on Keychron's fork.

NovaViper, 2023-10-01:
@adophoxia thank you for the clarification! Hopefully the teams over at Keychron and QMK manage to get something together and merge the changes.

MarioRicalde, 2023-10-15:
Since we have to rely on Keychron's fork, and based on the marketing material, it sounds only fair that the company puts work into getting the fork up to date with all those bug fixes from upstream.

Right?

adophoxia, 2023-10-17:
From what I've been told, the branch itself is based on Breaking Changes Q4 2022.

svenjacobs, 2023-11-06:
I contacted [Keychron support](https://keychronsupport.zendesk.com/hc/en-us/requests/new) with the following message

> Some months ago I bought a K3 Pro mainly because of the programmability via QMK.
Unfortunately I noticed afterwards that the keyboard is not supported by the main QMK repository. Instead I have to use the **bluetooth_playground** branch of your own [fork](https://github.com/Keychron/qmk_firmware/).
>
> What is even more unfortunate is that this branch is very outdated. It's several thousand commits behind the master branch. I kindly ask you to keep this branch in sync with master or merge it back into master. I would like to benefit from latest features and bug fixes of QMK.
> 
> As I said I mainly bought this keyboard because of QMK support. Since it is advertised as such, I expect to be able to use the latest QMK version. I'm a bit disappointed that this is not the case. There is also an [issue](https://github.com/Keychron/qmk_firmware/issues/145) about this. It would be nice if some Keychron dev could respond there.
>
> Thank you very much!

and got the following answer on Nov 2:

> We will pass it to our team and will take that into consideration. Have a nice day!

gericho, 2023-11-06:
I agree on every point, I will send an email regarding this to them too. 

> I contacted [Keychron support](https://keychronsupport.zendesk.com/hc/en-us/requests/new) with the following message
>
>
> and got the following answer on Nov 2:

tmas, 2023-12-31:
Since the issue with getting these changes merged seems to be solely based on Bluetooth license issues, maybe it could be worth creating non-bluetooth implementations of each keyboard? I bought a Pro board because of the keycaps/switches it came with and don't care about Bluetooth, so I'd be happy to run regular QMK without Bluetooth support. If I wanted to use the Bluetooth functionality later, I could always flash back to the stock firmware or build QMK myself based on Keychron's fork.

Unfortunately while I am a software developer, I mostly work on web projects and have _very_ limited knowledge of how QMK works. At some point I might attempt this myself, but whatever I come up with might not be worth releasing to the world.

K-Fuchs, 2024-01-08:
>  If I wanted to use the Bluetooth functionality later, I could always flash back to the stock firmware or build QMK myself based on Keychron's fork.
> 
Well, yes.
You can try my compiled binary or compile it yourself using the source code:

[https://github.com/K-Fuchs/q1_pro_via_no_bluetooth](url)

However, i'm unsure how it will impact the battery in the long run because of the battery controller and such. So i reccomend disconnecting it. And as always, do it on yout own risk and no warranties from my side at all.

But technically speaking, everything except bluetooth (duh) and the battery controller works.

Rokin05, 2024-03-05:
Just a short message to say that I'm in the same boat and that i'm shedding a little tear as i see that nothing is really moving forward.

svenjacobs, 2024-03-06:
I 100% agree! It is really sad that no Keychron representative has commented on this issue since it was created in July 2023 and that we're left alone with our concerns.

This was my first and last Keychron product 😡

gericho, 2024-03-06:
I agree with everything, It will be the last Keychron product for me as well. (K15 PRO)

adophoxia, 2024-03-06:
Said it before and I'll say it again. Their code is too custom to be merged into upstream QMK. The `wireless_playground` branch is more up to date with upstream, so that should leave little to talk about how it's old as heck to deal with.

svenjacobs, 2024-03-06:
[Related issue](https://github.com/Keychron/qmk_firmware/issues/217)

NovaViper, 2024-06-05:
Go away, Stalebot. This issue is very much so still important and is a pressing one.

adophoxia, 2024-08-28:
> Sorry if this is off-topic but is there an easy way to pull the latest QMK updates from the main QMK repo into a locally-cloned Keychron QMK repo?

No because it doesn't work like that. In order for Keychron to make wireless work on QMK for these boards, they made both, code changes of the repo and used code that the QMK team would not accept as it is too custom for them to do so.

nickshanks347, 2024-08-28:
> No because it doesn't work like that. In order for Keychron to make wireless work on QMK for these boards, they made both, code changes of the repo and used code that the QMK team would not accept [...]

Gotcha ok. Shame to be honest. Like others I expected full QMK compatibility as advertised by Keychron.

adophoxia, 2024-08-28:
> Gotcha ok. Shame to be honest. Like others I expected full QMK compatibility as advertised by Keychron.

What do you mean by "full QMK compatibility"? The boards _**do**_ use actual QMK firmware. Just not like as if it came from the main QMK repo.

svenjacobs, 2024-08-28:
> Just not like as if it came from the main QMK repo.

Right, but that is what customers expect when they read "QMK" and see the QMK logo on Keychron's product websites and packaging. My main concern is that I cannot just pull the newest version of QMK from the original repo and flash it onto my keyboard 😢

adophoxia, 2024-08-28:
For what it's worth, I've actually been trying to get the `k6_pro` code to compile on a branch based off of upstream `master`, but using the changes Keychron made for wireless to work. So far, only 1 error is stopping me from compiling successfully.

42Craft, 2024-10-14:
Sad to see that Keychron is not fixing this. This is not what you expect when you buy a QMK keyboard.

tboutaour, 2024-11-22:
Nov 2024. Another K2 Pro user who is upset about not hearing anything from keychron about this.

svenjacobs, 2025-02-21:
Still relevant!

mous16, 2025-03-05:
Still very relevant

Dawnflash, 2025-06-23:
Yeah please keep this branch up to date. The products like Q1 Pro are still relatively recent and this feels like abandoning them completely which doesn't reflect well on your ability to support your premium products.

rubienr, 2026-04-15:
I wouldn't care much if it is merged at least without wireless support. To be frank, i dislike that feature and never use it (it was never a reason so buy, rather a counter argument).

But since the OP started 2023 and three years later Keychron didn't move a pinky tho do anything about it, i want to make it clear to Keychron:

THAT  IS MY LAST KEYCHRON I HAVE EVER BOUGHT - Thank you Keychron, hello competitors!

WhyNotHugo, 2026-09-27:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Thbhfrgju