Keychron/qmk_firmware #146: [Bug] Some keys do not register when a text field is first selected (k12 pro)

kuokuo123, 2023-07-22:
### Describe the Bug

With the k12 pro's current qmk source code, under bluetooth mode with RGB off, some keys do not register after a text field is selected. This bug occurs when all the below conditions are satified and can be reproduced consistently:

1. flash a firmware compiled from k12 pro's current qmk source code, modified or unmodified.
2. unplug the usb wire and use bluetooth connection only
3. turn off RGB lighting
4. select a text field in any program on your screen (this is an important step)
5. press backspace (or other keys mentioned below) and it won't register

Keys that are not registering in this situation include backspace, backslash, enter, right shift and right control. The issue occurs only when using the firmware compiled from k12 pro's qmk source code, which means if you flash the stock firmware (v 1.0.0) downloaded from keychron's official website the issue is gone.

Notably, if you plug in the usb wire, or turn on RGB lighting, or do not select a text field beforehand, the keys will register as usual and as expected.

### Keyboard Used

keychron/k12_pro/ansi/rgb

### Link to product page (if applicable)

https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k12_pro

### Operating System

linux & windows 10

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

None

lokher, 2023-07-23:
can you try changing https://github.com/Keychron/qmk_firmware/blob/02a4691a88dbf3ce67ded5e863a65c7efdc6acaf/keyboards/keychron/k12_pro/matrix.c#L56

to 
```c
    for (i = DIRECT_COL_NUM; i < (MATRIX_COLS - DIRECT_COL_NUM); i++) {
```

kuokuo123, 2023-07-23:
Can confirm that changing this line fixes the issue. Now every key registers as expected. Thank you, lokher!

closed 2023-08-09 by lokher