Keychron/qmk_firmware #157: [Feature Request] Release files for K11 Pro

rayslinky, 2023-08-07:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Please release the files for QMK support of the K11 Pro. I would like to either write a custom keymap with leader macros or compile Vial for it and use combos. Either way, it is not connecting to Via at the moment.

lokher, 2023-08-09:
Here it is: https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k11_pro

You have to manully load json files in VIA to make it connected with VIA.

rayslinky, 2023-08-09:
Thanks for the update. Via still didn't seem to work for me from Chrome on Linux, even with the udev rules.

slabua, 2023-09-03:
> Thanks for the update. Via still didn't seem to work for me from Chrome on Linux, even with the udev rules.

@rayslinky I have to do `sudo chown <your_username>:<your_group> /dev/hidraw*`, refresh the page in chrome and then it works.

closed 2023-10-02 by rayslinky