Keychron/qmk_firmware #163: [Bug] [keychron K8 pro]mouse btn not functional

hodcarrier, 2023-08-23:
### Describe the Bug

use via maping some key to MOUSE_BTN1 (KC_MS_BTN1), and press that key, no functional.

### Keyboard Used

keychron k8 pro

### Operating System

windows

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2023-08-24:
Have you tried updating firmware?
https://www.keychron.com/blogs/archived/k8-pro-factory-reset-and-firmware-flash

hodcarrier, 2023-08-24:
not yet, will i lost warranty if I upgrade the keyboard?

hodcarrier, 2023-08-24:
[1] https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k8_pro/firmware
[2] https://www.keychron.com/blogs/archived/k8-pro-factory-reset-and-firmware-flash
I have tried both [1] and [2], the [2] works good. the [1] can not connect to via app.
is the source code of [1] in the git Repo? could you give me the tag/sha/version number in the git repo?
thanks very much.

4 layer and 15 macro is too few, I want to add some more layer based on the version that know works.

lokher, 2023-08-25:
> is the source code of [1] in the git Repo? could you give me the tag/sha/version number in the git repo?

Yes, use latest commit plz.

> 4 layer and 15 macro is too few, I want to add some more layer based on the version that know works.

VIA support max 15 macro, but it can be coded with QMK to use more.

lokher, 2023-08-27:
> I have tried both [1] and [2], the [2] works good. the [1] can not connect to via app.

Firmware of `bluetooth_playground` is VIA v3, while firmware from our website is VIA v2. Only v2 json file is on the VIA server, so we have to manually load the v3 json file when use firmware of VIA V3.

https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k8_pro/via_json

PeterMortensen, 2024-03-13:
> VIA support max 16 macro

It isn't limited to 16. And the number is set/determined by QMK.

For instance, adding this in config.h (in the appropriate place) will increase the number of macros in Via to 42:

    #define DYNAMIC_KEYMAP_MACRO_COUNT 42

The overhead is one byte per macro. Thus, in this example, the required (emulated) EEPROM space will be increased by 26 bytes.

I am typing this on a Keychron V6 with 42 Via macros, including one, M25 assigned to ***Fn*** + ***Ctrl*** (right Ctrl).