Keychron/qmk_firmware #165: [Bug]

Tianle-Leviathan, 2023-09-01:
### Describe the Bug

It does not compile if disable NKRO

### Keyboard Used

k1 pro

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I am trying to apply the apple fn patch and find that the code does not compile if set NKRO_ENABLE = no. The conditional compilation part is not done properly.

Tianle-Leviathan, 2023-09-01:
`RAW_EPSIZE` is missing due to NKRO_ENABLE = no. I find out that in my case (k1pro), it is 32.

lokher, 2023-09-02:
Please add `#include "usb_descriptor.h"` to `factory_test.c` and `ckbt51.c`, and change 

https://github.com/Keychron/qmk_firmware/blob/5ca4be13ed154ef22bdec8274b11b9b6777a10f3/keyboards/keychron/bluetooth/bluetooth.c#L104 to 
```c
#if defined(NKRO_ENABLE) && defined(BLUETOOTH_NKRO_ENABLE)
```

Tianle-Leviathan, 2023-09-02:
I did some hacky thing to work around this. I comment out the define of `BLUETOOTH_NKRO_ENABLE` and add `#define RAW_EPSIZE 32` in `ckbt51.h`. It is nice to see the propery fix.

closed 2023-09-02 by Tianle-Leviathan