Keychron/qmk_firmware #166: [Bug] Matrix last column keys sometimes do not register when using Bluetooth with RGB off (K11 Pro)

slabua, 2023-09-02:
### Describe the Bug

Perhaps this issue might be related to #146.
I just bought few days ago the K11 pro and I am still trying different use cases to verify the problem. I noticed the problem while playing a video and realised that the right arrow, along with the other keys belonging to the last column of the wiring matrix, did not register.

Keyboard runs the stock firmware that came shipped with, hardware switch is set to Windows mode.
I have slightly customised the layers 1, 3, 4 using VIA. Although I believe it is not related, I am attaching the json configuration file here, just in case.

Soon after turning off RGB with fn1+capslock, if not immediately, the keys of the last column of the matrix stop registering.
If any other key is pressed, the last column keys start working again for a very short while.
If any of usb mode or RGB light is in use, there seem to be no problem in registering the keys.

[k11procustom.zip](https://github.com/Keychron/qmk_firmware/files/12504343/k11procustom.zip)

### Keyboard Used

K11 Pro

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k11-pro-alice-layout-qmk-via-wireless-custom-mechanical-keyboard

### Operating System

Fedora

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2023-09-04:
Please update the firmware from below link
https://www.keychron.com/blogs/archived/k11-pro-factory-reset-and-firmware-flash

The new firmware should have fixed this issue.

slabua, 2023-09-04:
@lokher thanks, is the new firmware v1.01_20230815? Source code is actually from August 2nd. I bought the keyboard 5 days ago.
Yesterday I followed the instructions in #146 and it seems to have fixed the issue, although I haven't thoroughly tested it yet.

lokher, 2023-09-04:
The firmware also fixed encoder issue as PR #162 intend to do.

slabua, 2023-09-04:
@lokher thanks I am trying right now and it seems to be working so far.
I will close the issue for now, thanks again!

closed 2023-09-04 by slabua