Keychron/qmk_firmware #169: Keychron K Pro Series Bluetooth Firmware?

focus-mt, 2023-09-08:
Are there source codes of Keychron K Pro Series Bluetooth Firmware?
Thank you.

kaorihinata, 2023-09-08:
The source code for Keychron's K series keyboards, as well as basically all other keyboards that are currently advertised as supporting QMK/VIA, but that may not have been submitted upstream yet, are in the `bluetooth_playground` branch of the repository located [here](https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron).

lokher, 2023-09-10:
Sorry, the Bluetooth firmware source code cannot be opened due to the NDA signed with the chip factory.

focus-mt, 2023-09-11:
> The source code for Keychron's K series keyboards, as well as basically all other keyboards that are currently advertised as supporting QMK/VIA, but that may not have been submitted upstream yet, [...]

Ok. Got it. Thank you.

focus-mt, 2023-09-11:
> Sorry, the Bluetooth firmware source code cannot be opened due to the NDA signed with the chip factory.

Ok. Got it. Thank you.

But the keyboard name is shown as **_"Keychron K1 Pro"_** when I update version 1.32 on this URL:https://www.keychron.com/pages/keychron-k-pro-series-bluetooth-firmware.
Before the update it was shown as **_Keychron K1 Pro_**.

focus-mt, 2023-09-11:
> Sorry, the Bluetooth firmware source code cannot be opened due to the NDA signed with the chip factory.

Ok. Got it. Thank you.

But the keyboard name is shown as "Keychron K1 Pro" when I update to version 1.32 on this URL:https://www.keychron.com/pages/keychron-k-pro-series-bluetooth-firmware.
Before the update it was shown as Keychron K1 Pro.
Now it just looks a little bit curious. Is there any bluetooth firmware update plan for Keychron K Pro Series?

focus-mt, 2023-09-11:
> Sorry, the Bluetooth firmware source code cannot be opened due to the NDA signed with the chip factory.

Ok. Got it. Thank you.

But the keyboard name is shown as **_"Keychron K1 Pro"_** when I update to version 1.32 on this URL:https://www.keychron.com/pages/keychron-k-pro-series-bluetooth-firmware.
Before the update it was shown as **_Keychron K1 Pro_**.
Now it just looks a little bit curious. Is there any bluetooth firmware update plan for Keychron K Pro Series?

sencagri, 2023-09-13:
> Sorry, the Bluetooth firmware source code cannot be opened due to the NDA signed with the chip factory.

So why you are advertising as QMK supported?

svenjacobs, 2023-11-06:
> Sorry, the Bluetooth firmware source code cannot be opened due to the NDA signed with the chip factory.

@lokher I'm sorry, but you advertise these keyboards with QMK functionality, which is an open source firmware, but then cannot release the source code for these keyboards? 🤔

lokher, 2023-11-06:
> > Sorry, the Bluetooth firmware source code cannot be opened due to the NDA signed with the chip factory.
> 
> @lokher I'm sorry, but you advertise these keyboards with QMK functionality, which is an open source firmware, but then cannot release the source code for these keyboards? 🤔

The source code of main MCU is opened, almost all QMK functionalities are runned on it.

closed 2024-03-07 by github-actions[bot]