Keychron/qmk_firmware #172: [Bug] when pressing <R-ALT> key, R-ALT and L-CTRL gets recognized

nnako, 2023-09-16:
### Describe the Bug

I am currently experiencing that when pressing the right ALT key (`<R-ALT>` or `KC_RALT`), not only the right ALT key but at the same time also the left CONTROL key (`<L-CTRL>` or `KC_LCTL`) gets recognized and triggered by the keyboard.

I am using a K8 PRO ISO RGB (German) keyboard and tried this using the "[usevia.app](https://usevia.app)" application on the KEY TESTER pane.

[image]

questions

- Does anybody know why the keyboard(s) behave like this?
- Is it an unspoken law to operate maybe only German keyboards like this?
- Is there a way to modify this low-level configuration to be able to recognize only a `KC_RALT` when a `<R-ALT>` is pressed, and no additional `KC_LCTL`?

### Keyboard Used

Keychron K8 Pro RGB (German)

### Operating System

Windows 10

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2023-09-18:
Can you show me the keymap in VIA?

nnako, 2023-09-21:
Sorry. Don't understand. How do I get to the "keymap in VIA"? Is "VIA" the web application I open when issuing https://www.caniusevia.com/docs/supported_keyboards in the browser?

nnako, 2023-09-25:
any hint? any help on this?

lokher, 2023-09-26:
Can you try holding fn + j + z  these three keys for 4 seconds to factory reset the keyboard?

nnako, 2023-09-26:
yes, I did that. no change regarding the issue. just some configuration changed which resulted in the delay time until keyboard sleep to be lower than before. after starting "usevia.app", the delay is normal again. but the issue is still there.

lokher, 2023-09-27:
Can you share the screenshoft as beflow of all layers with me? 
[image]

nnako, 2023-09-27:
[image]
[image]
[image]
[image]

the only thing I see is that on German keyboards (as mine) the `<Y>` and `<Z>` are exchanged. I use the "Windows" switch position which activates "Layer 2", as I have read.

nnako, 2023-09-27:
here, I took a small video describing what you asked me to do:

https://github.com/Keychron/qmk_firmware/assets/5845602/54db5b39-8159-4318-90a1-4bba7d88d6ea

lokher, 2023-09-28:
Can you test it on another PC? Looks like a OS issue

nnako, 2023-09-28:
Indeed. I checked on an English Linux machine. And here, the test (without selecting "Test Matrix") shows that when pressing `<R-ALT>` only a `KC_RALT` gets detected. No `KC_LCTL`. Which is exactly what I had expected.

But what could be the reason for this? Is this a problem (or maybe it is considered as a special "feature") of only German versions of Windows OS? Would there be ways to deactivate this "feature" within the Windows OS?

nnako, 2023-10-05:
Is there anything I can do to move forward, solving that issue locally? Any hints?

lokher, 2023-10-08:
Can you test in safe mode of Windows? If it works fine in safe mode, try disabling unnecesary startup option.

nnako, 2023-10-12:
are there no further hints available? where could I find more information about where to further check relevant operating system configurations?

lokher, 2023-10-13:
> are there no further hints available? where could I find more information about where to further check relevant operating system configurations?

Have you tried safe mode?

mbartlett21, 2024-01-12:
@nnako 
If you are on Windows, the AltGr key (which Windows maps right Alt to on some keyboard layouts) is passed to applications as Ctrl+Alt rather than a separate keycode.