Keychron/qmk_firmware #174: OpenRGB Integration?

NovaViper, 2023-09-27:
### Issue Description

Hey I'm wondering how I can enable OpenRGB integration on the Keychron K8 Pro. I downloaded the newest prebuild firmware that's available on this repo but OpenRGB still says it's not enabled in the make settings.

adophoxia, 2023-10-02:
> I downloaded the newest prebuild firmware that's available on this repo.
Can you elaborate more on this and show where you got said firmware from? I'm pretty sure that the only way this would be possible is if you'd compiled the firmware that has OpenRGB support enabled, but then again, you would have to know how to correctly merge Keychron's `bluetooth_playground` branch with `Kasper24/QMK-OpenRGB:openrgb` fork, which would already be a PITA to begin with.

NovaViper, 2023-10-02:
> > Can you elaborate more on this and show where you got said firmware from? 
These ones here: https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k8_pro/firmware

I got the one for the ANSI RGB version of the K8 pro

adophoxia, 2023-10-02:
That's not how that works, meaning what I said still holds true.

NovaViper, 2023-10-02:
Ah yeah.. I haven't a faintest how to merge the two different firmware (especially since the QMK firmware that has OpenRGB support is a completely different branch off the main one).

NovaViper, 2024-01-07:
This issue is still relevant! It doesn't need to be closed

gericho, 2024-01-09:
agree!

HybridAir, 2024-04-10:
Please don't close this, it's still relevant! While the lighting can be manually adjusted using VIA, having the option to use OpenRGB  would open up lots of options for reactive and live lighting control.

tebowy, 2024-07-11:
I hate autoclose bots

tebowy, 2024-10-11:
> This issue has been automatically marked as stale because it has not had activity in the last 90 days. It will be closed in the next 30 days unless it is tagged properly or other activity occurs. [...]

stfu

NovaViper, 2025-01-10:
No this needs to stay open

NovaViper, 2025-04-11:
Can we please get this issue labeled as on-hold at least? So it doesn't get closed randomly?