Keychron/qmk_firmware #175: [Bug] K11 Pro. Some keys do not wake up keyboard in the Bluetooth mode

hedgss, 2023-09-30:
### Describe the Bug

**Keyboard**: K11 Pro (K11P-H1)
**Connection**: Bluetooth
**Configuration:** a new keyboard with the default configuration.

**Description:**
When the keyboard is connected via Bluetooth, it goes to save power mode after some delay (Lights are turned off).
If I press some keys e.g. space it will wake up and work properly.
But if I try to use some keys e.g. Knob it is nothing happens.

The list of the keys that have this behaviour:
- Knob
- Del
- Home
- All arrows.

If I press space, then the Knob works again.
It is pretty inconvenient because if I want to adjust sound I need to press another key and only then I can use the Knob.

### Keyboard Used

K11 Pro

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k11-pro-alice-layout-qmk-via-wireless-custom-mechanical-keyboard?variant=40531261718617

### Operating System

Windowss 11 Pro

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2023-10-08:
Sorry for the trouble, please update the firmware, new firmware should fix it:
https://www.keychron.com/blogs/archived/k11-pro-factory-reset-and-firmware-flash

hedgss, 2023-10-29:
I have updated the firmware. I flashed k11_pro_ansi_knob_rgb_v1.01_20230815.bin to my keyboard.
I confirm that it was fixed.

Thanks a lot for the fast fix!

closed 2023-10-29 by hedgss