Keychron/qmk_firmware #176: [Feature Request] OpenRGB Support

Phara0h, 2023-10-02:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Just got the Q6 Pro and thought it had OpenRGB support out of the box since it shows that on the OpenRGB website that it supports QMK boards. However this is not the case. Looks like this fork is missing all the OpenRGB functionality on mast and the branch i need "Bluetooth playground". Can we get this functionality made for these keyboards? This is your guys "Pro" line, but its software is below falling behind your non "Pro" line which is inside main QMK.

gericho, 2023-11-10:
Agree, here a K15 pro RGB user, expecting the same.

svenjacobs, 2023-11-21:
Also see [this issue](https://github.com/Keychron/qmk_firmware/issues/145). The support of their Bluetooth keyboard is currently lacking, unfortunately.

closed 2024-03-21 by github-actions[bot]