Keychron/qmk_firmware #179: [Bug] VIA can't fetch V3 definitions for Keyboard K8 Pro V3 pre-compiled firmware; received invalid protocol version from device

NovaViper, 2023-10-04:
### Describe the Bug

I downloaded the [precompiled firmware for the RGB Ansi version of the K8 Pro that's in the repo](https://github.com/Keychron/qmk_firmware/blob/bluetooth_playground/keyboards/keychron/k8_pro/firmware/keychron_k8_pro_ansi_rgb_via.bin) and flashed it but whenever I open VIA now, I get these errors when loading up the keyboard in the program (on the webapp and on desktop)
```
19:15:51.654
Fetching v3 definition failed
Device: Keychron Keychron K8 Pro
Vid: 0x3434
Pid: 0x0280

19:16:00.895
The device must be opened first.
Device: Keychron Keychron K8 Pro
Vid: 0x3434
Pid: 0x0280

19:16:00.972
Received invalid protocol version from device
Device: Keychron Keychron K8 Pro
Vid: 0x3434
Pid: 0x0280
```

### Keyboard Used

Keychron K8 Pro

### Operating System

NixOS 23.11 (unstable)

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2023-10-08:
Only VIA v2 json is merged to VIA github, v3 json need to be manaully loaded. 
The json file is in ([via_json](https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k8_pro/via_json)) 

Please refer to below link to use it:
https://www.keychron.com/blogs/archived/how-to-use-via-to-program-your-keyboard

NovaViper, 2023-10-27:
Hey I'm closing the issue since this has to be done manually

closed 2023-10-27 by NovaViper