Keychron/qmk_firmware #180: [Bug]  stuttering sound

workpk43, 2023-10-05:
### Describe the Bug

I am using three devices:
1. K15 Pro, K2 V2
2. Mouse Master Mx 3s
3. Headphone SoundPeats Air 3
I have noticed that when I connect the K15 Pro, it causes intermittent audio stuttering after a while of listening. However, I don't encounter this issue when using the K2 V2. Could you please help me troubleshoot this?

Device Specs:
K15 Pro firmware version 1.02
Mac OS Ventura 13.2.1
Mouse Master mx 3s

Thx.

### Keyboard Used

K15 Pro

### Operating System

Mac OS

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

SmetDenis, 2023-10-05:
I have the same issue with K15 Pro.

workpk43, 2023-10-10:
Initially, keychron support has provided this file for testing. It was found that initially there was no longer this problem.

[Keychron.zip](https://github.com/Keychron/qmk_firmware/files/12860919/Keychron.zip)

closed 2023-10-10 by workpk43