Keychron/qmk_firmware #197: [Bug] q1 pro compile error after the jis_encoder commits

norbertc, 2023-10-30:
### Describe the Bug

While compiling for Q1 Pro:

`Compiling: keyboards/keychron/bluetooth/bat_level_animation.c                                       [OK]
Compiling: keyboards/keychron/bluetooth/rtc_timer.c                                                 [OK]
Compiling: keyboards/keychron/q1_pro/q1_pro.c                                                      In file included from ./lib/chibios/os/hal/include/hal.h:299,
                 from platforms/chibios/platform_deps.h:18,
                 from quantum/quantum.h:18,
                 from keyboards/keychron/q1_pro/q1_pro.h:19,
                 from keyboards/keychron/q1_pro/q1_pro.c:17:
keyboards/keychron/q1_pro/q1_pro.c: In function 'keyboard_post_init_kb':
keyboards/keychron/q1_pro/q1_pro.c:177:43: error: 'encoder0_pad_cb' undeclared (first use in this function); did you mean 'encoder_pad_cb'?
     palSetLineCallback(encoders_pad_a[0], encoder0_pad_cb, NULL);
                                           ^~~~~~~~~~~~~~~
./lib/chibios/os/hal/include/hal_pal.h:994:31: note: in definition of macro 'palSetLineCallback'
     palSetLineCallbackI(line, cb, arg);                                     \
                               ^~
keyboards/keychron/q1_pro/q1_pro.c:177:43: note: each undeclared identifier is reported only once for each function it appears in
     palSetLineCallback(encoders_pad_a[0], encoder0_pad_cb, NULL);
                                           ^~~~~~~~~~~~~~~
./lib/chibios/os/hal/include/hal_pal.h:994:31: note: in definition of macro 'palSetLineCallback'
     palSetLineCallbackI(line, cb, arg);                                     \
                               ^~
In file included from ./lib/chibios/os/hal/include/hal_pal.h:174,
                 from ./lib/chibios/os/hal/include/hal.h:299,
                 from platforms/chibios/platform_deps.h:18,
                 from quantum/quantum.h:18,
                 from keyboards/keychron/q1_pro/q1_pro.h:19,
                 from keyboards/keychron/q1_pro/q1_pro.c:17:
keyboards/keychron/q1_pro/q1_pro.c: In function 'matrix_scan_kb':
keyboards/keychron/q1_pro/q1_pro.c:190:26: error: 'CKBT51_RESET_PIN' undeclared (first use in this function); did you mean 'CRC_CR_RESET_Pos'?
             palWriteLine(CKBT51_RESET_PIN, PAL_LOW);
                          ^~~~~~~~~~~~~~~~
./lib/chibios/os/hal/ports/STM32/LLD/GPIOv3/hal_pal_lld.h:437:67: note: in definition of macro 'pal_lld_writegroup'
   uint32_t w = ((~(uint32_t)(bits) & (uint32_t)(mask)) << (16U + (offset))) | \
                                                                   ^~~~~~
./lib/chibios/os/hal/include/hal_pal.h:601:37: note: in expansion of macro 'pal_lld_writepad'
 #define palWritePad(port, pad, bit) pal_lld_writepad(port, pad, bit)
                                     ^~~~~~~~~~~~~~~~
./lib/chibios/os/hal/include/hal_pal.h:716:33: note: in expansion of macro 'palWritePad'
 #define palWriteLine(line, bit) palWritePad(PAL_PORT(line), PAL_PAD(line), bit)
                                 ^~~~~~~~~~~
./lib/chibios/os/hal/include/hal_pal.h:716:61: note: in expansion of macro 'PAL_PAD'
 #define palWriteLine(line, bit) palWritePad(PAL_PORT(line), PAL_PAD(line), bit)
                                                             ^~~~~~~
keyboards/keychron/q1_pro/q1_pro.c:190:13: note: in expansion of macro 'palWriteLine'
             palWriteLine(CKBT51_RESET_PIN, PAL_LOW);
             ^~~~~~~~~~~~
At top level:
keyboards/keychron/q1_pro/q1_pro.c:134:13: error: 'encoder_pad_cb' defined but not used [-Werror=unused-function]
 static void encoder_pad_cb(void *param) {
             ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [.build/obj_keychron_q1_pro_jis_encoder_default/keyboards/keychron/q1_pro/q1_pro.o] Error 1
make: *** [keychron/q1_pro/jis_encoder:default] Error 1
Make finished with errors
`

### Keyboard Used

Q1 Pro

### Operating System

MacOS

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

`Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/norbert/kb-fw/qmk_firmware
Ψ Detected macOS 14.1 (Apple Silicon).
Ψ Git branch: bluetooth_playground
Ψ Repo version: 0.14.29
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found`

mikaraunio, 2023-10-31:
Also seeing this. ef55878 is the last commit that builds succesfully for me, 76937cb and newer error out as above.

norbertc, 2023-11-01:
fixed on commit 29bf1b3

closed 2024-02-01 by lokher