Keychron/qmk_firmware #198: q1v2 special, which keyboard model should I use

diegoroccia, 2023-10-30:
Hello I have a Q1v2 special edition, and I am not sure which model I should use. Looking at the hardware, it looks like it corresponds to a Q1 (v1) model, is this correct? is there any difference?

lokher, 2023-11-10:
```Ultra-low-power Arm Cortex-M4 32-bit STM32L432 (128KB Flash)

*The special edition (Orange Yellow, Champagne Gold, Red, Retro, Green, and Purple version) is equipped with MCU ATMEGA32U4.
```

Better way is to check the bootloader with keyboard in DFU mode in QMK toolbox, if the yellow text include "STM32", plz use `q1v2`, otherwise use `q1v1`.

diegoroccia, 2023-12-27:
@lokher can you help me understand which keyboard should I set to be able to build a firmware for my Q1V2 Special edition? I am trying with all combinations but it's not working

lokher, 2023-12-28:
> @lokher can you help me understand which keyboard should I set to be able to build a firmware for my Q1V2 Special edition? I am trying with all combinations but it's not working
> 
> ```
> Ψ Bootloader Connected: atmel-dfu: ATmega32U4
> ```

use q1v1 please

diegoroccia, 2023-12-28:
from the main QMK branch? because when I tried, it almost bricked my keyboard

```
git clone https://github.com/qmk/qmk_firmware.git qmk
qmk config user.keyboard=keychron/q1v1/ansi_encoder
qmk config user.keymap=default
qmk flash
```

the keyboard became unresponsive for several minutes and then nothing worked anymore. Luckily I had my fork with a working firmware so I could flash it back, but I am unsure what's breaking it

this is the branch that works, and my keyboard definition is here: https://github.com/diegoroccia/qmk_firmware/tree/diego/keyboards/keychron/q1v2special

I would like to drop it and use the official definitions, but I need some help here please

diegoroccia, 2024-02-14:
could you answer my question on which model to use for my Keychron Q1V2 Special?

closed 2024-02-14 by diegoroccia