Keychron/qmk_firmware #201: [Bug] Keyboard not awakening with knob volume dial rotation K15 pro

gericho, 2023-11-06:
### Describe the Bug

When keyboard is in sleep mode, rotating the knob is doing nothing instead of waking up the keyboard properly.

### Keyboard Used

K15 pro

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k15-pro-alice-layout-qmk-via-wireless-custom-mechanical-keyboard

### Operating System

Windows 11 

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2023-11-10:
Please update the firmware to fix it
https://www.keychron.com/blogs/archived/k15-pro-factory-reset-and-firmware-flash

gericho, 2023-11-10:
Solved with the keychron linked firmware (keychron website). I assume the branch here on github is updated too. Closing this.