Keychron/qmk_firmware #202: Implement bluetooth battery / energy service

ispiropoulos, 2023-11-14:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

I have recently bought the K3 Pro and was quite surprised to see that there is no reporting of battery percentage to the host system.

I am talking about the percentage shown on other devices (eg like the Nuphy air60 mechanical keyboard) shown here:
[image]

Needless to say, the `fn+b` macro does "output" an approximation of the battery left but it is debatable if the user experience is any good, waiting for several seconds for the lights to glow:

https://github.com/Keychron/qmk_firmware/assets/2028642/cc2364a7-da59-4bb6-bc98-e6a07cecf86a

Now, this has been discussed a lot, from a quick search one can see that this is a popular request:

- https://www.reddit.com/r/Keychron/comments/17ktzud/check_battery_percentage_on_macos/
- https://www.reddit.com/r/Keychron/comments/qgss3s/battery_percentage/
- https://www.reddit.com/r/Keychron/comments/174g52d/k1_pro_get_battery_percentage_to_show_in/
- https://www.reddit.com/r/Keychron/comments/l2hatc/is_there_a_way_to_see_current_battery_percentage/
- https://www.reddit.com/r/Keychron/comments/14oi40c/is_it_possible_to_display_battery_percentage/
- https://www.reddit.com/r/Keychron/comments/i1ooel/is_there_a_way_to_know_how_much_battery_left_on/
- https://www.reddit.com/r/Keychron/comments/rsg8fn/battery_percentage/
- https://www.reddit.com/r/Keychron/comments/gvk7e5/check_keychron_battery_life/
- https://www.reddit.com/r/Keychron/comments/hf0z1g/how_to_check_battery_charge_level_on_k1_v4_in/
- https://www.reddit.com/r/Keychron/comments/nx9mu9/check_battery_level_with_k3/
- https://www.reddit.com/r/Keychron/comments/yqsx9x/why_doesnt_my_k3_show_battery_life_in_win_mac/

Can this item be planned for implementation? A wirelesses keyboard that comes with macOS support by default shouldn't be missing this feature.

svenjacobs, 2023-11-21:
This would be a great feature! Unfortunately I have the impression that the K3 Pro is [not very well supported](https://github.com/Keychron/qmk_firmware/issues/145) by Keychron 😢

michael-letcher, 2024-01-18:
Same with my Q6 Pro. I'm tempted to see if I can create a PR for it.

huangsam, 2024-04-19:
Would like to see this become a reality for the Q Pro line 😄 - hopefully the issue can be labeled properly so that it's kept alive!

closed 2024-08-18 by github-actions[bot]