Keychron/qmk_firmware #203: [Bug] Q11 Keyboard Not Starting After Reboot with Self-Compiled Firmware

AmmarAbouZor, 2023-11-21:
### Describe the Bug

I am experiencing an issue with the Q11 keyboard, where it fails to start after a computer reboot when using a firmware compiled by myself. This behavior is not observed when using the provided pre-built firmware.

I found the same issue [same issue](https://github.com/qmk/qmk_firmware/issues/21686) reported in the mainstream repository, and a [solution](https://github.com/qmk/qmk_firmware/issues/21686#issuecomment-1664572148) was discussed there which solved the problem for me. I believe applying the same fix to the Keychron fork would benefit users who compile their firmware.

Thank you for addressing this issue, and I appreciate your efforts in maintaining and improving the Keychron firmware.

### Keyboard Used

Keychron Q11

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q11-qmk-custom-mechanical-keyboard

### Operating System

Windows

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lededev, 2023-12-23:
You need install qmk_toolbox drivers, including stm32-dfu.
And flash factory firmware backup to keyboard in dfu mode.
To enter dfu mode, keyboard turn off, hold ESC key (or switch button under space bar) and turn keyboard on.

AmmarAbouZor, 2024-01-17:
@lededev Thanks for your suggestion, however I don't want to flash the pre-built firmware, instead I want to build the firmware with my customization for home-row mods like tapping_term and combos, and I've figured out how to do solve the problem. I just wanted to point out that you need to insert the linked [solution](https://github.com/qmk/qmk_firmware/issues/21686#issuecomment-1664572148)  in your source code for the Q11 if this is the right one, so other users like me don't have to search that solution.

AmmarAbouZor, 2024-07-14:
The fix has been added to QMK mainstream repo. This can be closed now

closed 2024-07-14 by AmmarAbouZor