Keychron/qmk_firmware #204: [Feature Request]  Add k13 Pro Iso source code

benoit2600, 2023-11-23:
### Feature Request Type

- [ ] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi, 

I've found the ansi source code for the K13 pro in the bluetooth_playground branch, but the ISO version is missing. 

Could you add it ?

Thanks. 

Regards,

Benoit

oyvindma, 2024-01-21:
Please add it, as I need to remap my keys. Alternatively, are there any other way?

closed 2024-05-21 by github-actions[bot]

PeterMortensen, 2024-08-22:
The source code for the two K13 Pro ISO variants has [now been released](https://github.com/Keychron/qmk_firmware/issues/290#issuecomment-2304419929) (2024-08-19).