Keychron/qmk_firmware #207: bluetooth playground upsteam

Dialgatrainer02, 2023-12-26:
### Issue Description

is it possible to upstream bluetooth playground branch or is it stuck on an older qmk release 
 if i cant be up streamed how would you update it locally

elfakyn, 2023-12-27:
++ing this one. there are bugs in this branch related to RGB state when host suspend/unsuspend that are not present in the 1.01 firmware, as well as other different behaviors. Can we have the latest code for the K-series keyboards? Thanks!

lokher, 2024-01-11:
Some Pro keyboards are in `wireless_playground` branch, based on recent upstream.
https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron

Dialgatrainer02, 2024-01-11:
thank you so much  wireless playground was exactly what i needed thank you

closed 2024-01-11 by Dialgatrainer02