Keychron/qmk_firmware #208: i need V1 MAX source code !!!

DeGrey, 2023-12-29:
### Issue Description

when is available about  keychron v1 max source code

adophoxia, 2024-01-10:
[Ahem...](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/v1_max)

DeGrey, 2024-01-11:
well, thanks a lot.
it is certanly v1 max code.but it can't compile because lack of following files:
```
qmk_firmware/lib/printf/src/printf/*
qmk_firmware/lib/lufa/LUFA/*
qmk_firmware/lib/chibios/*
```
i copied them from qmk master branch. and compile has passed.

lokher, 2024-01-11:
> well, thanks a lot. it is certanly v1 max code.but it can't compile because lack of following files:
> 
> ```
> qmk_firmware/lib/printf/src/printf/*
> qmk_firmware/lib/lufa/LUFA/*
> qmk_firmware/lib/chibios/*
> ```
> 
> i copied them from qmk master branch. and compile has passed.

Did you run "git submodule update"?

DeGrey, 2024-01-11:
yes
i've tried update submodule and build default keymap flash to my keyboard.
It's work normally.
Thanks for your reply.

closed 2024-01-11 by DeGrey