Keychron/qmk_firmware #216: [Bug] K3 Max not responding with latest firmware from github after key combination

xserk, 2024-01-12:
### Describe the Bug

I compiled the bin file for the K3 Max using QMK MSYS both 1.8.0 and 1.7.2. I made no modifications to the source code before compiling. In both cases, after flashing the keyboard it seemed to work fine. But after I press they key combination Fn + Lights Button to turn off the lights, the keyboard stops responding and wont take any further commands.

### Keyboard Used

keychron/k3_max/ansi/white:default

### Operating System

Windows 10

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/Haris/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ QMK MSYS version: 1.7.2
Ψ Userspace enabled: False
Ψ Git branch: wireless_playground
Ψ Repo version: 0.14.29
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest wireless_playground: 2024-01-11 17:56:25 +0800 (74b08459bc) -- Added Q1 Max; Updated V1 Max
Ψ - Latest upstream/master: None
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: None
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9ba)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [X] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2024-01-12:
Fixed, plz fetch and try again.

xserk, 2024-01-12:
Alright thanks! I will try it. Just a quick question, I wanna try changing the debounce value cause I got a lot of double presses with a few switches. Do I just need to change the debounce value found inside info.json from 20 to 30 for example? Or should I define debounce somewhere else? Thank you.

xserk, 2024-01-12:
Ok, tested it. Problem with Fn + Lights Button is fixed, but there are two other issues I found out.
1) Fn + B doesnt show battery status,
2) Keyboard is no longer recognized in VIA App even after reloading the json you provide on products page.

lokher, 2024-01-12:
1. Fn+B desen't work when usb cable is connected.
2. VIA is not supported in default keymap, via keymap is for it.
`make keychron/k3_max/ansi/white:via`

For the debouce, try setting `debounce_type` to `sym_defer_g`, and `deboulce` to 5, this is the default setting. Increase `debounce` value if issue still exists.

xserk, 2024-01-12:
Thank you, all works as expected.

commit 8aeaeaef (2024-01-12): Changed LED_MATRIX_SHUTDOWN_ENABLE to LED_MATRIX_DRIVER_SHUTDOWN_ENABLE to fix #216
commit 85213783 (2024-01-12): Changed LED_MATRIX_SHUTDOWN_ENABLE to LED_MATRIX_DRIVER_SHUTDOWN_ENABLE to fix #216
closed 2024-01-12 by xserk