Keychron/qmk_firmware #217: Question about the current software state

frostbyte-ninja, 2024-01-16:
### Issue Description

Hi,

I really like the Keychron hardware, but I'm a bit disappointed with how the "Open Source" promise is being managed here. I've come across issues that make the current state not quite acceptable for keyboards advertised as QMK-compatible.

**Issue 1: Fork of QMK**
Keychron has its own fork of QMK for licensing or other reasons, which isn't ideal but not a deal-breaker either. I've heard this is being worked on with the QMK team, so I guess this issue is already being addressed.

**Issue 2: Code Location**
The Keychron-specific code is stored in various branches with names containing the word "playground." For example, the K3 Pro product page links to a branch called `bluetooth_playground`, while the Lemokey L3 page links to `wireless_playground`.

- `wireless_playground`: 7 commits ahead of, 45 commits behind qmk/qmk_firmware:master
- `bluetooth_playground`: 278 commits ahead of, 3260 commits behind qmk/qmk_firmware:master
- `playground`: 28 commits ahead of, 635 commits behind qmk/qmk_firmware:master
- `old_playground`: This branch is 443 commits ahead of, 2388 commits behind qmk/qmk_firmware:master

How's anyone supposed to figure out the right branch to use here? I had to contact support, which pointed me to the `wireless_playground` branch since I was looking for the Lemokey L3 firmware. Additionally, it seems that this one also contains more recent K3 Pro firmware??

**Issue 3: Up-to-date Status**
Those branches tend to get stale and deviate a lot from the master branch over time, causing missing bug fixes and features. This is pretty bad for keyboards that have "QMK" in the headline of their product description. For something like that, I'd expect a software state as close to the QMK-master as possible.

**Suggestion:**
As long as this fork is necessary, I'd like to see a single "Keychron"-specific branch (e.g., "keychron-master") set as the default, containing all the source for Keychron boards not currently supported in QMK upstream. This branch should also get frequent updates from the QMK-master branch.

Can we have something like that?

lokher, 2024-01-19:
> ### Issue Description
> Hi,
>
> I really like the Keychron hardware, but I'm a bit disappointed with how the "Open Source" promise is being managed here. I've come across issues that make the current [...]

**Branches**

`playground` branch is mainly for wire keyobard model, it's created for these reasons:
- keyboard source code hasn't been merged
- bugs of QMK upstream or related submodules refer to core changes, there will take long time, may be half a year or even more, and some have no plan
- some factory test mode code that upstream may not allowed

`bluetooth_playround` is for K Pro and Q Pro series BT and USB dual mode keyboards, and `wireless_playground` contains all Max series keyboards besides Pro series. 

 `wireless_playground`  is recommended for all Keychron keyboard with QMK and bluetooth support.

Though `bluetooth_playground` is not recommanded, it fixed bricking issue which caused by lacking of NMI handling triggered by ECCD error on flash reading. Latest upstream seems fixed it either, but haven't been verified since it's difficult to reproduce ECCD error. 

`playground` can get frequent update since it rarely touch core files, merging is easier, but bluetooth related branches won't since it touch more core files.

For better experience, I think we can create a guide branch as default branch to guide to these branches.

**Commits**

We only care about effective core commits, none core commits are mostly keyboards/keymaps we don't care.
There are 315 open and 19,564 close PRs so far,  among these PRs,  89 open and 3,068 closed PRs labeled `core`, among core PRs, 607 closed PRs are  unmerged. Some PRs are not our keyboard haredware related, etc MCU, led drivers and so on, and  some of the core PRs just rename macro ....

With our keyboard  hardware, I don't think we miss import keyboard feature in `bluetooth_playground` branch though is 3260 commits behind, please open an issue if there is new feature missing in these branchs.

Rokin05, 2024-03-06:
Thanks, I'm a bit confused about the bluetooth_playround and wireless_playground branches.

For example for the K3 Pro :
-  The firmware on this page corresponds to the wireless_playground branch: https://www.keychron.com/blogs/archived/k3-pro-factory-reset-and-firmware-flash
- The firmware on this page corresponds to the bluetooth_playround branch : https://www.keychron.com/pages/keychron-k-pro-series-bluetooth-firmware

_bluetooth_playround is an optional branch for those who have a problem between the keyboard and the K Pro series headphones._

is that right ?

Alternatively, concerning the difficulties explained which necessitate maintaining a fork of qmk_firmware, 
I don't know how much work this can represent, but wouldn't it be possible to consider proposing a simple firmware without the wireless part, which could be more easily integrated on the qmk/qmk_firmware project side?

It's maybe a bit extreme but it would allow users to at least have the choice between following the main qmk/qmk_firmware project with only the wired part or staying on Keychron/qmk_firmware to get wireless function.

lokher, 2024-03-07:
> bluetooth_playround is an optional branch for those who have a problem between the keyboard and the K Pro series headphones.

No, It's just a fallback due to it have been tested, but `wireless_playground` hasn't.

>Alternatively, concerning the difficulties explained which necessitate maintaining a fork of qmk_firmware,
I don't know how much work this can represent, but wouldn't it be possible to consider proposing a simple firmware without the wireless part, which could be more easily integrated on the qmk/qmk_firmware project side?

If we PR without wireless part, users will confuse, and our support team and qmk upstream will need to spend time on explain it, it will waste evertyone's time. 

Maintenance work includes comparison, modification and testing. 
First, look through the change log and PRs and check core PRs content.
Second, make change to code.
Finally, test the new firmware, this include DQA test and daily use, it takes 1 or 2 weeks.

Another problem is we have our developing work on new product, we can only deal with it when we are not busy.

svenjacobs, 2024-03-07:
> Another problem is we have our developing work on new product, we can only deal with this when we are not busy.

Well, that's questionable. Existing products should also be supported for a long term. That means good customer support for me.

frostbyte-ninja, 2024-03-07:
> **Branches**
>
> `playground` branch is mainly for wire keyobard model, it's created for these reasons:
>
>     * keyboard source code hasn't been merged
>
>     * bugs of QMK upstream or related submodules [...]

This still sounds quite convoluted to me. Why are all these tasks being done on different branches? I don't see anything similar in QMK upstream, and they support a lot more hardware.

> Another problem is we have our developing work on new product, we can only deal with this when we are not busy.

Does this mean product maintenance is only performed when there is no development for new products? That's a prime example of why people desire upstream support for their hardware.

lokher, 2024-03-07:
> > Another problem is we have our developing work on new product, we can only deal with this when we are not busy.
> 
> Well, that's questionable. Existing products should also be supported for a long term. That means good customer support for me.

It will be supported for a long term, but we won't update as frequent as QMK upstream.

svenjacobs, 2024-03-11:
> Users don't need to know much about it. Most uses will look up source code in QMK upstream, the source code it's there if mostly if the keyboard is wired only. And our support team will tell users to use 'wireless_playground` branch if they ask Keychron support team for wileless kebyoard.

How come other keyboard manufacturers are able to merge their code into the official QMK repository? Is it because of Bluetooth or is your code somehow incompatible with the QMK repository?

Either way, do you plan to merge your code into the QMK repo in the long run?

bbucko, 2024-03-11:
> So the question remains, will keychron be merged upstream or is it going to remain an incompatible code source with QMK?

I generally agree with your statement about incompatibility and ideally we would see Keychron's QMK to be compatible with the upstream repo but I want to point out that keychron **is** compatible with QMK (just not the most recent version) and **is** open source (source code is available). So both claims are **technically** correct.

bbucko, 2024-03-11:
I disagree that the customers are misled. I think that some of the assumptions made by the customers are over-reaching - I did the same mistake when I assumed that wireless dongle provides the same capabilities as the wired USB connection (I wanted to do OS-detection via inspection of USB packets). I was mistaken and it's my fault.
But.. The keyboards are marketed as having support for QMK and are using the QMK's logo which means that Keychron is obliged to create a PR with keyboard's configuration when the keyboard is shipped (as per https://qmk.fm/powered/). Maybe this is the way to force them to update the fork because clearly they are not doing it. I'm not sure if anyone can enforce this obligation but I'm not a lawyer ;)

exstntlmsnthrp, 2024-03-12:
> Maybe this is the way to force them to update the fork because clearly they are not doing it. I'm not sure if anyone can enforce this obligation but I'm not a lawyer ;)
>

Yes, both points are technically correct here. Alas, it comes down to one of the perennial questions associated with open source licensing - enforcement. If someone had enough money, they could even be taken to court in a locale where they legally conduct business and a shrewd enough attorney could argue to enforce this obligation. But, in my opinion, I think this puts the foci on the letter of the law and less on the spirit, as I personally interpret much of this discussion. 

Would I like to see keychron do so willingly and of their own accord? Definitely, as it appears many here do as well. Do I think they will do so? I don't know, but I'm skeptical, especially after their few responses.

If I had the time, I'd clone both repos and attempt it myself. But then it would neither be official and not likely accepted upstream.

svenjacobs, 2024-06-11:
This is still an issue. Could maintainers please label it accordingly so that the bot doesn't bother us?

kitsunekyo, 2024-07-26:
i'm mostly at odds with the lack of stable / up2date documentation on how to properly use the repo and branches.

i wasted a weekend trying to comb together information on how to actually use QMK with my k3 pro. which is frustrating as heck. a wiki page with how to to get started and which branches to use would really help

eldorel, 2024-11-03:
I'm going to drop a message here in support of the points made above. 

I've bought several of the keychron wired boards, and just received my first wireless one. 
I *specifically* bought the Q3 pro instead of the Max because it had been out for  almost a year, and I wanted to avoid waiting for a merge like I did with my office board. 
I didn't realize that the additional variant code wasn't in the QMK repository. 

The fact that I cannot build directly off of my existing fork from the main branch for this is annoying enough that I am currently shopping for a different wireless keyboard. 

When I account for my time, it's cheaper to just replace this one than it is to work out all of the coding problems that I expect to encounter.

hanselsin, 2024-11-07:
> I'm going to drop a message here in support of the points made above.
>
> I've bought several of the keychron wired boards, and just received my first wireless one. I _specifically_ bought the Q3 pro [...]

I have been looking at the K13 Max and want to rebuild my own firmware. Did the QMK installation, built the example keyboard firmware according to the QMK tutorial.
Then I cloned the wireless-playground branch, but still haven't been able to build the firmware successfully. Just wonder after these few months if anyone is able to build a customized firmware (to my interest it is K13 Max) out of the Keychron repo?

I didn't realize forking the repo can actually causing so much troubles though.

PeterMortensen, 2024-11-14:
> if anyone is able to build a customized firmware

Yes, that is not a problem. But the standard QMK instructions do ***not*** work. [Special setup of QMK](https://www.reddit.com/r/Keychron/comments/1cxnqtu/comment/l55cx8r/) is required.

The compile commandline is something like:

    qmk clean # To make changes (if any) to .json files take effect
    qmk compile -kb keychron/k13_max/iso/rgb -km via

Result:

    66082	  10222	keychron_k13_max_iso_rgb_via.bin

Though the .bin file takes up about 95 KB (about 50% more): 

    98388 Nov 14 01:29 keychron_k13_max_iso_rgb_via.bin

That was with 7CBA260 (2024-10-15).

### References

* [K13 Max product page](https://www.keychron.com/products/keychron-k13-max-qmk-via-wireless-custom-mechanical-keyboard). 
* [K13 Max source code](https://github.com/Keychron/qmk_firmware/blob/wireless_playground/keyboards/keychron/k13_max). Note: In [Keychron's fork](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron) and in that fork, in [Git](https://en.wikipedia.org/wiki/Git)&nbsp;branch ***"wireless_playground"*** (not the default&nbsp;branch). No matter the Git&nbsp;branch, for example, ***"wireless_playground"***, it requires [special setup of QMK](https://www.reddit.com/r/Keychron/comments/1cxnqtu/comment/l55cx8r/) (the standard QMK instructions and many other guides will ***not*** work (because they implicitly assume the main QMK&nbsp;repository and a particular Git branch)). [Source&nbsp;code&nbsp;commits](https://github.com/Keychron/qmk_firmware/commits/wireless_playground/) ([RSS&nbsp;feed](https://github.com/Keychron/qmk_firmware/commits/wireless_playground.atom). Latest:&nbsp;2024-11-12).

frostbyte-ninja, 2024-11-14:
> Yes, that is not a problem. But the standard QMK instructions do not work. [Special setup of QMK](https://www.reddit.com/r/Keychron/comments/1cxnqtu/comment/l55cx8r/) is required (essentially, adding three extra parameters to 'qmk setup' will make it work).

Arguable. This is information I do not expect to be scattered all over the Internet, or buried 20 comments deep in some random GitHub issue.

PeterMortensen, 2024-11-14:
> Arguable.

Well, Keychron is not providing any help. They just ***dump*** here source code here.

bam80, 2024-12-13:
Could we, as a community, merge the code upstream?

appetrosyan, 2025-01-15:
> Could we, as a community, merge the code upstream?

As a community, I would appreciate that we pushed keychron to remove the false advertising, and only post QMK compatibility **if** and _when_ its code is actually merged. 

So far the only review that mentions this very real problem is mine, 
https://www.keychron.com/products/keychron-k13-pro-qmk-via-wireless-custom-mechanical-keyboard

42Craft, 2025-01-16:
I share the same sentiments expressed in this topic. I hope Keychron can step up their game.

vinc6, 2025-02-26:
I echo most of the comments made above. 
Got my K2 HE months ago. Before pull trigger I had a brief reading on Keychron official website and its github source code page, and I thought that I can follow docs from QMK and make customization for my keyboard. With helps from the community and after spent 2-3 weekends spare times to learn that playground branches things (in my case `hall_effect_playground`), I am able to make my custom firmware now, at this point I didn't really care about K2 HE is QMK official support or not.

Then I bought another Keychron keyboard, Q60 Max for my office use. Hardware-wise, both are excellent keyboards, but I realize I have to deal with two branches (`wireless_playground` and `hall_effect_playground`). To this moment I don't see any articles or guide from Keychron.

I hope Keychron could at least merge all their playgrounds to let say `keychron_playgrounds`, and provide their own documentations for customize firmware, and guide for people like me who owns 1 or more Keychron keyboards.

thematrixincendies, 2025-03-24:
There should be a single branch for all keychron keyboards which should be regularly backported from master, i.e. it should be rebased onto master every 2 weeks or something.

appetrosyan, 2025-03-26:
> There should be a single branch for all keychron keyboards which should be regularly backported from master, i.e. it should be rebased onto master every 2 weeks or something.

Well ideally, there should be _no_ keychron branch, the code is merged upstream, and nobody needs to faff around in different branches. 

For a company that advertised QMK support, making sure that it is supported as well as firmware for other keyboards is the _bare minimum_. 

Even if such a "well-maintained" branch were to be in good standing at all times, it is not a substitute. If the official position of Keychron is that QMK are being facetious and not merging the code... then perhaps they should have considered advertising QMK support **if** and **when** the merge happened. 

This is not some FOSS project where there's a disclaimer about warranty. I bought a keyboard that does not have an advertised feature. And before you say "but there's a branch", I'll ask you if you consider a company advertising IP68 water resistance to be lying to you if years after releasing the product they have a **failed test** and no evidence of work done.

svenjacobs, 2025-03-26:
Well said, although at this point I would be fine with a single branch of this fork that is regularly rebased onto QMK's upstream `main`.

The whole situation is really frustrating because Keychron's hardware is actually pretty good.

thematrixincendies, 2025-03-26:
> Well ideally, there should be _no_ keychron branch, the code is merged upstream, and nobody needs to faff around in different branches.
>
> For a company that advertised QMK support, making sure that [...]

I completely agree.

> Well said, although at this point I would be fine with a single branch of this fork that is regularly rebased onto QMK's upstream `main`.
> 
> The whole situation is really frustrating because Keychron's hardware is actually pretty good.

Exactly, it wouldn't be ideal, but far better than the current situation.

kitsunekyo, 2025-03-27:
This exact reason is why I will ditch my K3 Pro for the next best thing with official QMK support, although I really love the keyboard. 

The fact that Keychron keeps marketing QMK support on keyboards, while I, as a customer, have to jump through hoops (like riddiculous branching strategies) is infuriating.

bam80, 2025-03-27:
FYI: [ZMK](https://github.com/Keychron/qmk_firmware/issues/344) should be far more manageable, I think we could to make an upstream PR ourselves if Keychron bails out.

42Craft, 2025-03-28:
> Well ideally, there should be _no_ keychron branch, the code is merged upstream, and nobody needs to faff around in different branches.
>
> For a company that advertised QMK support, making sure that [...]

I totally agree.

thematrixincendies, 2025-03-31:
Also, I just noticed something interesting: https://www.keychron.com/pages/firmware-and-json-files-of-the-keychron-qmk-keyboards
There have been updates to version 1.1.0 for almost all Q series keyboards (just ANSI, though, not ISO), which is definitely not reflected by this repo.

PeterMortensen, 2025-04-02:
> ...definitely not reflected by this repo.

I have [noticed that too](https://www.reddit.com/r/Keychron/comments/1jl7cl7/comment/mkyykq4/?context=3).

thematrixincendies, 2025-04-02:
> > ...definitely not reflected by this repo.
> 
> I have [noticed that too](https://www.reddit.com/r/Keychron/comments/1jl7cl7/comment/mkyykq4). I am tracking several branches using their RSS feeds (using Thunderbird's reader) and haven't noticed such changes. Though some of them _may_ have been behind feature flags for a long time, for example, the NKRO part ([seems to be forced now](https://www.reddit.com/r/Keychron/comments/1j092s2/comment/ml0wljn/)).

I think these are the NKRO changes: https://github.com/qmk/qmk_firmware/commit/6aa63c3c636b97fc06ab3e9f1d7b99e7931a738f and https://github.com/qmk/qmk_firmware/commit/e5e57f406e4281f3a9e61afdb6c3e7d323c0cf7b
These are actually the reason why I built the firmware from source, because these changes are not reflected in the released v1.0.0 firmware (I use the ISO variant). These two commits fixed macros not working for me with NKRO enabled.

But I don't see anything in the repo pointing to a per key RGB setting, which is mentioned in the changelog of v1.1.0.
Neither do I understand why they would release v1.1.0 for the ANSI layout, but not for the ISO layout. Those are just layout changes, the majority of those feature additions should have happened in non-layout specific source files.

PeterMortensen, 2025-04-04:
[We have been teased](https://www.reddit.com/r/Keychron/comments/1jpylkl/comment/mlb3o6m/)...

PeterMortensen, 2025-04-12:
It was my impression that branch ***"wls_2025q1"*** was mostly for the QMK support for the K series, e.g., [K8 V2](https://keychron.ca/products/keychron-k8-qmk-wireless-mechanical-keyboard-version-2) (and perhaps to update to a newer QMK version).

For example, where are [the debounce changes](https://www.reddit.com/r/Keychron/comments/1jl9b63/comment/mlkggdq/) (now dynamic, settable through the web configurator)? I would expect some changes in [folder */quantum/debounce*](https://github.com/Keychron/qmk_firmware/tree/wls_2025q1/quantum/debounce)*.

PeterMortensen, 2025-05-10:
[Noted again](https://www.reddit.com/r/Keychron/comments/1kj93de/q6_max_firmware_and_qmk/).

jose-elias-alvarez, 2025-05-10:
It's really disappointing to buy a premium keyboard with clearly advertised QMK support only to find that it comes with a *ton* of caveats. Is there any reason why Keychron is not on [QMK's list of license violators](https://docs.qmk.fm/license_violations)?

> Typical non-compliance includes:
> 
>  - Not providing any source code
>  - Providing "crippled" source code, such as a wired-only firmware for a wireless-capable board

appetrosyan, 2025-05-12:
> It's really disappointing to buy a premium keyboard with clearly advertised QMK support only to find that it comes with a ton of caveats. Is there any reason why Keychron is not on [QMK's list of license violators](https://docs.qmk.fm/license_violations)?

I think this is exceptionally generous for Keychron. Their keyboards are OK. 

But yes. This is something that should be fixed in the violators.

ThrRip, 2025-05-12:
> It's really disappointing to buy a premium keyboard with clearly advertised QMK support only to find that it comes with a _ton_ of caveats. Is there any reason why Keychron is not on [QMK's list of [...]

Because there isn't a license violation. The full source code is provided in this repository.

Yes, wireless keyboards from Keychron are running a fork of QMK, but the fork has the same license as `qmk/qmk_firmware`.

ThrRip, 2025-05-12:
I figured I'd mention this as no one did: https://github.com/qmk/qmk_firmware/issues/25117#issuecomment-2786602672

> They have made extensive changes to core QMK code, especially for their wireless offerings. As such we havent been in a position to accept their contributions.

I guess Keychron has tried to merge into upstream, but the QMK team didn't have enough resources to review the massive amount of changes regarding wireless support.

> Well said, although at this point I would be fine with a single branch of this fork that is regularly rebased onto QMK's upstream main.

The [`wls_2025q1`](https://github.com/Keychron/qmk_firmware/tree/wls_2025q1) branch is probably their take on this. I've just built and flashed it onto my new V6 Max, and everything works fine.

svenjacobs, 2025-05-13:
> The [`wls_2025q1`](https://github.com/Keychron/qmk_firmware/tree/wls_2025q1) branch is probably their take on this. I've just built and flashed it onto my new V6 Max, and everything works fine.

As it is now it's an absolute mess. There are several branches and no documentation. How am I supposed to know which branch I should use for my model? What Keychron at least should do is provide a single branch with all their changes or use their `main` and regularly rebase it onto QMK's `main`. I don't understand why they make it so complicated?

ThrRip, 2025-05-13:
> How am I supposed to know which branch I should use for my model?

[The first reply on this issue](https://github.com/Keychron/qmk_firmware/issues/217#issuecomment-1899558528) still applies. I agree that this is far from clear documentation. But for now:

- `playground` for wired-only models
- `wireless_playground` for all the other models
- `wls_2025q1` for those who want a newer QMK version

> What Keychron at least should do is provide a single branch with all their changes or use their `main` and regularly rebase it onto QMK's `main`. I don't understand why they make it so complicated?

TBH I have no idea 🙃 My guess is that those changes made to bring 2.4 GHz support to QMK make this a painful process. And they just decided to chill on a stable version. Open-source with longevity is not easy.

asunekants, 2025-05-14:
> Open-source with longevity is not easy.

That would be a reasonable excuse if Keychron was a volunteer project, and not a commercial entity with actual paid devs. The expectation is very different.

appetrosyan, 2025-05-14:
So here's a subtle point that I think is a bit lost. 

When I was shopping for a keyboard, I had the choice between a Q3 Pro and K13 Pro. These both advertise QMK support. There is no indication that the support for K13 was yeeted at QMK. If they need to follow a procedure to get the code merged, then you only advertise support for QMK **after** it gets merged. 

The merge isn't even in progress, I have no certainty that it would ever get merged. In fact, more likely it will be abandoned by Keychron. 

For a piece of code provided to me for free; yeah, there's the disclaimer about non-warranty. For a physical good? Yeah, this is false advertising.

jose-elias-alvarez, 2025-05-15:
> > Open-source with longevity is not easy.
> 
> That would be a reasonable excuse if Keychron was a volunteer project, and not a commercial entity with actual paid devs. The expectation is very different.

Absolutely. On top of that, Keychron is profiting from the work of *actual volunteers* and using "open-source" and "QMK" as selling points when the actual experience of using QMK on the keyboards is not up to par. That may not constitute an actual license violation, but it certainly violates both the spirit of the license and consumer expectations.

PeterMortensen, 2025-05-27:
Re *"That may not constitute an actual license violation"*: It is a license violation to not release the source code at the same time as the (physical) product.

Users clearly don't have access to the source code.

Or has it been released? Can someone point to some ***particular*** source code file where some of the new features in the early 2025 firmware release have been implemented?

PeterMortensen, 2025-06-03:
Now something happened in branch ***"[wireless_playground](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron)"**** (2025-06-03. [C90496](https://github.com/Keychron/qmk_firmware/commit/C9049679ACA08A079EF05D1C1F86D73FC6FC922C)):

 * *"[Added snap click, per-key/mixed RGB, custom debounce, wireless config, etc.](https://github.com/Keychron/qmk_firmware/commit/c9049679aca08a079ef05d1c1f86d73fc6fc922c)"*

I don't about know about [the next one](https://github.com/Keychron/qmk_firmware/commit/8b525cb77025589b5b0b97c5cba9be67d12b392a). Why would V3 Max be special? Was it forgotten in the previous commit.

appetrosyan, 2025-06-07:
These very keen businessmen took down a three star review that I left two years ago on the model that I bought which is affected. I will not be buying a keychron. And since they cannot be trusted to leave an unbiased review on their platform, I will do my best to convey the information directly in a blog post. 

How is this not grounds for a class action lawsuit is beyond me.

PeterMortensen, 2025-06-17:
From [another merge attempt](https://github.com/qmk/qmk_firmware/pull/22751#issuecomment-2404173539) (NuPhy, not Keychron), on 2024-10-10:

> *"This is one of those [PRs](https://en.wiktionary.org/wiki/pull_request#Noun)
> that has enough wrong with it
> that reviewers are almost nonexistent&mdash;the PR checklist
> published by QMK outlines a significant number of things
> that need addressing before anything would normally
> proceed, and reviewers generally leave PRs alone if things
> in the checklist aren't handled. Stuff like introducing
> new lighting systems and duplicating QMK code effectively
> render this PR as "unmergeable" in its current form, yet
> you've seemingly got the expectation that we should ignore
> our requirements just for your favourite board.*
>
> *The PR checklist is there so we, as maintainers, don't
> need to continually outline the same problems people keep
> making in minute detail. It's not up to us to inform PR
> submitters which clauses they don't comply with.*

And:

> *Additionally, there are discussions going on with other
> manufacturers regarding how to integrate
> wireless&mdash;doing things as they're done in this
> PR has no future
> within QMK. Manufacturers have been told to maintain
> their own forks in the meantime."*

PeterMortensen, 2025-07-11:
Re *"Any reason it is not rebased"*: A guess would be too many (breaking) API changes. On the simpler end of the scale are renaming of keycodes, renaming of files and folder, and [removal of folders](https://www.reddit.com/r/olkb/comments/1ezxvwt/comment/lkhpen0/). There is also the on-going move to [data-driven configuration_](https://docs.qmk.fm/data_driven_config#data-driven-configuration).

Tymon3310, 2025-09-17:
the [2025q3](https://github.com/Keychron/qmk_firmware/tree/2025q3) branch was created. for now only one commit changing to (new) keychron's chibios branch

PeterMortensen, 2025-10-30:
Re *"the 2025q3 branch was created ... only one commit"*: This later changed.

But I *think* it is only intended for the wired-only Keychron keyboards, in particular the new "8K" keyboards, [C1 Pro 8K](https://github.com/Keychron/qmk_firmware/tree/2025q3/keyboards/keychron/c1_pro_8k), [C2 Pro 8K](https://github.com/Keychron/qmk_firmware/tree/2025q3/keyboards/keychron/c2_pro_8k), and [C3 Pro 8K](https://github.com/Keychron/qmk_firmware/tree/2025q3/keyboards/keychron/c3_pro_8k)

appetrosyan, 2025-11-02:
Advertise QMK support. Have non-existent QMK support. Make vapid promises. 

> On discord dev (@lokher ) said "yes, all the keyboards will be in one single branch"

Ask them why they remove negative reviews for keyboards that falsely advertise support for QMK. 

[image]

I left a three star review. And a one-star review as well. Why don't they show up?

codebam, 2025-11-03:
@appetrosyan Where did you post the review? I want to make sure it shows up eventually. Otherwise I'll be reporting Keychron to Consumer Protection. For this and the false advertising of QMK support.

@lokher (not your problem, but legal might care)

PeterMortensen, 2025-11-08:
Re *"Have non-existent QMK support"*: What do you mean? For which keyboard? Has the source code not been released for it?

Is it about the source code corresponding to [the 2025 Keychron keyboard main firmware updates](https://www.reddit.com/r/Keychron/comments/1j092s2/comment/ml0wljn/)? Yes, the source code release of May/June 2025 was only a partial release. Fortunately, it is possible to live without those changes (they can all be done directly in QMK).

Or about the (effective) QMK version?

For the K Pro series and Q Pro series, branch [wls_2025q1](https://github.com/Keychron/qmk_firmware/tree/wls_2025q1/keyboards/keychron) gets you a QMK version of February 2025 (this year), and [it is known to work](https://github.com/Keychron/qmk_firmware/issues/396#issuecomment-3102239204). The same for the V Max, Q Max, and K QMK series.

Yes, it is more complex than it ought to be, and you may not get support for the latest version of QMK or [Vial](https://get.vial.today) support, but on the whole it is possible to work it work.

But Keychron ***really*** ought to release all source code for all keyboards (and in the future, ***in a timely manner***, instead of delayed up to one year or more), ***on&nbsp;their&nbsp;own&nbsp;accord***, without *[having&nbsp;to&nbsp;be&nbsp;prompted&nbsp;to&nbsp;do&nbsp;it](https://www.reddit.com/r/Keychron/comments/1ezuv5b/comment/lkpw7r5/)*.

Tymon3310, 2025-11-14:
they did it, look at the commit history https://github.com/Keychron/qmk_firmware/commits/2025q3/

PeterMortensen, 2025-11-20:
Re *"they did it"*: No, they started it. For example, in the Q HE and K HE series, only a single keyboard is currently included ([source code for the Q3 HE](https://github.com/Keychron/qmk_firmware/tree/2025q3/keyboards/keychron/q3_he). [Q3 HE product page](https://www.keychron.com/products/keychron-q3-he-qmk-wireless-custom-keyboard).).

thematrixincendies, 2026-01-18:
> Re _"2025q3 ... No, they started it"_: Since then, the source code was added for most of the [HE](https://www.tomshardware.com/news/how-hall-effect-switches-work,33105.html) keyboards (K HE series, Q HE series, L HE series, and P HE series), V1 Max, V6 Max, and K9 Max, but there still isn't a single representative for the Q Max series ([for example, the Q1 Max](https://www.reddit.com/r/Keychron/comments/1qb5sb1/comment/nz850on/)).

Exactly and even the [wls_2025q1](https://github.com/Keychron/qmk_firmware/tree/wls_2025q1) branch that does include the Q Max series is not up to date with the firmware released on the Keychron website!
[On their website](https://www.keychron.com/pages/firmware-and-json-files-of-the-keychron-qmk-keyboards) the firmwares provided as v1.1.0 and v1.1.1 (the latter btw only exists for ANSI, not for ISO, why???) is newer than what we get in the wls_2025q1 branch. I flashed the wls_2025q1 and it doesn't include advanced lighting options, nor the configurable debounce, so it is older/different.

PeterMortensen, 2026-02-17:
The source was added one week later.

epokuser, 2026-11-16:
still waiting for keychron k4 pro keyboard source code update....