Keychron/qmk_firmware #218: [Bug] BT audio stutter on windows11 with K2 Pro

thomaslagies, 2024-01-17:
### Describe the Bug

Whenever I use the Keychron K2 Pro via Bluetooth on my windows 11 PC, I get audio stutters with my BT Headset.
The Keychron is the only device, that causes this. Other bluetooth devices are fine to connect with windows. 

I'm running the latest windows update (23H2 Windows 11 Pro - Build 22631.3007) and the latest keychron firmware was flashed (https://www.keychron.com/pages/special-firmware-for-k2)

I don't have these issues on my Mac M1 for example. 
My Phone also works fine with Headset and Keyboard connected.

### Keyboard Used

Keychron K2 Pro QMK

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k2-pro-qmk-via-wireless-mechanical-keyboard

### Operating System

Windows 11 Pro

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

Logitech Software for mouse

lokher, 2024-01-18:
Trye this bluetooth firmware
[keychron_ckbt51_01.32-2.zip](https://github.com/Keychron/qmk_firmware/files/13970722/keychron_ckbt51_01.32-2.zip)

adophoxia, 2024-01-18:
@thomaslagies, you mention you have a **Keychron K2 Pro**, but in the 2nd paragraph of this post, you mentioned you flashed firmware for the [Keychron K2 V2](https://www.keychron.com/pages/firmware-for-k2-version-2), aka, the regular K2. Is that right?

thomaslagies, 2024-01-18:
> Trye this bluetooth firmware [keychron_ckbt51_01.32-2.zip](https://github.com/Keychron/qmk_firmware/files/13970722/keychron_ckbt51_01.32-2.zip)

Will try this the other day. Thanks

thomaslagies, 2024-01-22:
This seems to have fixed all the stutter!
Thanks mate. 
Also I found the post about audio stutter on the keychron website (via reddit)
How could I miss that for almost half a year?!

closed 2024-01-22 by thomaslagies