Keychron/qmk_firmware #223: Make error with Keychron V3 iso_encoder

Love-M, 2024-02-03:
### Issue Description

Hi,

I was following a guide to compile the qmk firmware for use with vial.

In qmk msys I ran make keychron/v3/iso_encoder:vial -

Im getting the following errors -

Compiling: keyboards/keychron/common/keychron_common.c                                             keyboards/keychron/common/keychron_common.c: In function 'encoder_pad_cb':
keyboards/keychron/common/keychron_common.c:117:5: error: implicit declaration of function 'encoder_interrupt_read' [-Werror=implicit-function-declaration]
  117 |     encoder_interrupt_read((uint32_t)param & 0XFF);
      |     ^~~~~~~~~~~~~~~~~~~~~~
keyboards/keychron/common/keychron_common.c: In function 'keyboard_post_init_kb':
keyboards/keychron/common/keychron_common.c:124:9: error: implicit declaration of function 'palEnableLineEvent' [-Werror=implicit-function-declaration]
  124 |         palEnableLineEvent(encoders_pad_a[i], PAL_EVENT_MODE_BOTH_EDGES);
      |         ^~~~~~~~~~~~~~~~~~
keyboards/keychron/common/keychron_common.c:126:9: error: implicit declaration of function 'palSetLineCallback' [-Werror=implicit-function-declaration]
  126 |         palSetLineCallback(encoders_pad_a[i], encoder_pad_cb, (void *)i);
      |         ^~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:376: .build/obj_keychron_v3_iso_encoder_vial/keychron_common.o] Error 1
Make finished with errors
make: *** [Makefile:416: keychron/v3/iso_encoder:vial] Error 1

I was wondering if this is an issue with my setup or with keychron_common.c file.

Thanks

Love-M, 2024-02-03:
I think this might be a question for the vial firmware github.

Love-M, 2024-02-03:
When compiling I copied the keychron playground branch keychron folder to the vial keychron folder replacing all files.

adophoxia, 2024-02-03:
Why use the files from `playground` when the V3 already exists in upstream QMK (qmk/qmk_firmware)?

Love-M, 2024-02-03:
The guide I was following said the playground branch as it was more up to date.

Love-M, 2024-02-03:
I followed the guide and didnt use the playground files and all good,  thanks.

closed 2024-02-03 by Love-M