Keychron/qmk_firmware #225: [Bug] VIA doesn't work on K11 Pro

hedgss, 2024-02-06:
### Describe the Bug

I followed [the guide](https://www.keychron.com/blogs/archived/how-to-use-via-to-program-your-keyboard) to configure Keychron K11 Pro, but I cannot authorize device.

VIA detects successfully the keyboard, but when I press connect, nothing happens.

### Keyboard Used

K11 Pro

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k11-pro-alice-layout-qmk-via-wireless-custom-mechanical-keyboard?variant=40531261718617

### Operating System

Windowss 11 Pro

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

 Chrome Version 121.0.6167.140 (Official Build) (64-bit)

adophoxia, 2024-02-07:
So you didn't load the JSON in the Design tab?

Sasuchisama, 2024-02-15:
@hedgss Do you use a usb-hub? Test a direct connection with the original cable.

hedgss, 2024-02-22:
Thanks for the answers!
Yes, I had both issue.
I connected the keyboard directly and uploaded the JSON and now it works.
Thanks a lot!

closed 2024-02-22 by hedgss