Keychron/qmk_firmware #227: how to repurpose the os dip switch k8 pro

Dialgatrainer02, 2024-02-08:
### Issue Description

i use linux and i have the dip switch set to win/android. i also have custom layers and removed the mac os ones from keymap.c. how to i make the dip switch change to one of my custom layers and not attempt to switch to mac os then fail. 
when i do switch it to mac os if i try to press tab a few of the buttons blink red. 

i tried setting my dip_switch_update_user() in my keymap.c and i get compilation errors to about factory_test.o

sjthearts, 2024-02-29:
rename dip_switch_update_user to orig_dip_switch_update_user in factory_test.c and rebuild.

Dialgatrainer02, 2024-03-10:
tried this and still errors it even errors when i comment out the whole function in factory test.c 
i have a k8 pro

closed 2024-07-10 by github-actions[bot]