Keychron/qmk_firmware #229: K10 pro + manual firmware build + Bluetooth

anenviousguest, 2024-02-12:
### Issue Description

Hi folks, K10 Pro owner here.

I have a K10 Pro with a stock firmware and wanted to configure VIA macros which uses [delays](https://github.com/the-via/releases/issues/24) whose support has been added to VIA long ago.

Sample macro code could look like:

```
{KC_LGUI}{500}mycoolsoftware{500}{KC_ENT}
```

However, after pairing VIA paired with my keyboard and trying to edit the macro code, I've got an error message from VIA saying that delays are not supported with current version of firmware.

Never mind, I built the firmware from the source code, trying both [wireless_playground](https://github.com/Keychron/qmk_firmware/tree/wireless_playground) as well as [bluetooth_playground](https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground) branches.

On its own, the firmware builds just fine, and also I can successfully flash it with QMK Toolbox. Also macro starts supporting delays, yay! 

[image]

_…However!!!..._

Seems like the firmware built from those branches does not fully support Bluetooth functionality - I still can pair the device anyhow, but the keyboard is not recognized as a _keyboard_ any longer, and the keystrokes do not work - tried with both `*playground` branches listed above. After flashing to the stock firmware, Bluetooth is working fine again. 

I researched the issues on this repo and found few mentions that Bluetooth-related code is not published due to it being NDA'ed - but does that really mean that the firmware built from the source code for K10 Pro is never going to support Bluetooth?

Or am I missing something?

Seems like a bummer to be forced to choose between Bluetooth and latest VIA support 😅

Thanks.

lokher, 2024-02-17:
Can you try the firmware https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k10_pro/firmware

anenviousguest, 2024-02-29:
BT actually works.

closed 2024-02-29 by anenviousguest