Keychron/qmk_firmware #230: Please add K3 V2   ( NOT PRO NOT MAX ) support

cuzrawr, 2024-02-13:
### Feature Request Type

- [X] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

Bought new K3 v2 ( K3E3 ) from China, and I not very happy with default keyboard settings. 

So i started to dig the internet, already sent e-mail to keychron support but they send me to discord (is that a joke?)

I found the sonix qmk, guys reverse it and make it works.

There sonix qmk project exist that can add QMK / Via support to the K3 v2
  but i don't want to lose my BT support, aslo default PWM settings are broken so my led are flickering, but with default keychron firmware are somehow avoiding this.
  
also  i don't care about bluetooth firmware, keep it closed source because as there mentioned somewhere its NDA, so don't open it. I don't need Bluetooth sources!

But ONLY thing what i want it this keymap:

[image]

Some stupid RGB button placement make me waste 2 weeks of research.

Its not possible to do with default firmware on K3 V2, so please add QMK and/or VIA support.

Its already working firmware but need officially support, here the link:

https://github.com/SonixQMK/qmk_firmware/tree/sn32_master/keyboards/keychron/k3

_For whos read this don't buy K3 v2 (not max / pro) its a disaster._

cuzrawr, 2024-02-16:
Okay for speedup your work, I deploy configs for Sonix QMK,
tested on K3v2RGB Optical

Here pull:

https://github.com/SonixQMK/qmk_firmware/pull/374

Please fix it by your own and add normal BT support.

lokher, 2024-03-04:
K3 is not QMK keyboard.

closed 2024-03-04 by lokher

cuzrawr, 2024-03-09:
> K3 is not QMK keyboard.
No, it's partially works on **Sonix QMK** fork (that i mentioned).
I even write own _matrix.c_ to for it, _bluetooth_ still not working. But that pain Not worth  **80$** .

**Keychron still selling this plastic scrap k3v2 on their website. I wouldn't recommend Keychron keyboards to anyone else.**

lokher, 2024-03-10:
> No, it's partially works on **Sonix QMK** fork (that i mentioned). I even write own _matrix.c_ to for it, _bluetooth_ still not working. But that pain Not worth **80$** .
>
> **Keychron still selling [...]

Yes, some keyboards can be ported to work with QMK by thirdparty, but regular K series is not advertised as QMK keyboard on Keychron product page.

MaluNoPeleke, 2024-11-28:
What about the new K3 v3?
It seems to officially support QMK: https://www.keychron.com/products/keychron-k3-qmk-wireless-mechanical-keyboard-version-3
Only downside: *The K3 Version 3 is currently compatible with Gateron/Keychron low-profile 2.0 mechanical switches only, as QMK does NOT support low-profile optical switches.