Keychron/qmk_firmware #233: [Bug] q6_max/ansi_encoder/ansi_encoder.c

naitoshedo, 2024-02-25:
### Describe the Bug

// Key Matrix to LED Index
The LED index is incorrect as there are 20 LEDs on the top row (Escape to X keycap)
21 keys on the second row
21 keys on the third row counting the Numpad Plus key
16 keys on the fourth row
17 keys on the fifth row (counting enter key on numpad)
13 keys on the sixth row

firmware wise in ansi_encoder.c these counts do not match the physical device. 

### Keyboard Used

keychron/q6_max/ansi_encoder

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q6-max-qmk-via-wireless-custom-mechanical-keyboard

### Operating System

Windows / Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/qmk/keychron/wireless
Ψ Detected Linux (Debian GNU/Linux 11 (bullseye)).
Ψ Userspace enabled: False
Ψ Git branch: wireless_playground
Ψ Repo version: 0.14.29
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9ba)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

Dudleydogg, 2024-02-25:
Thank you for Submitting

lokher, 2024-02-26:
it's not a bug. 
Usual matrix for 100% keyboard is 6 * 21, but we use 6 * 20 matrix due to lack of enough GPIOs, the right most column use unused matrix positions as below:
[image]

and the  physical position is here
[image]

the mapping aligns 
[image]

Dudleydogg, 2024-02-27:
Thanks for the confirmation / information

Dudleydogg, 2024-03-01:
what I notice is that even in VIA I can't program the last row of keys, no matter what layer you're on they do what they do. X - + "Enter" key are programmable.

lokher, 2024-03-01:
> what I notice is that even in VIA I can't program the last row of keys, no matter what layer you're on they do what they do. X - + "Enter" key are programmable.

Can you elaborate?