Keychron/qmk_firmware #238: [Bug] VIA not detecting Q5 Pro

lith-x, 2024-03-03:
### Describe the Bug

I flashed my Q5 Pro with the firmware from the `wireless_playground` branch, using the command `qmk compile -kb keychron/q5_pro/ansi_encoder -km via` in QMK MSYS then flashing using the QMK Toolbox. I coerced the VIA web app like how described in the guide [here](https://www.keychron.com/pages/how-to-factory-reset-or-flash-your-qmk-via-enabled-keychron-q5-pro-keyboard) on the Keychron website, but any key I tried to remap didn't seem to reflect on the keyboard, which led me to download the windows desktop app, where it clearly isn't picking up on it either in wired or wireless modes.

### Keyboard Used

keychron/q5_pro/ansi_encoder

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q5-pro-qmk-via-wireless-custom-mechanical-keyboard?variant=40492440977497

### Operating System

Windows 10

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: C:/Users/[user]/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ QMK MSYS version: 1.8.0
Ψ Userspace enabled: False
Ψ Git branch: wireless_playground
Ψ Repo version: 0.23.1
Ψ - Latest wireless_playground: 2024-03-02 11:41:28 +0800 (a56ef872dd) -- Add V6 Max
Ψ - Latest upstream/master: 2024-03-02 08:45:01 +1100 (1875659df0) -- CLI Speed improvements. (#23189)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-12-18 13:53:02 +1100 (9539f135d8) -- Remove obvious user keymaps, `keyboards/[0-9]*` edition. (#22691)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9ba)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Was directed here after finding the issue someone else was having over on the VIA github https://github.com/the-via/releases/issues/286

lokher, 2024-03-04:
Which layer did you change in VIA? Layer 2 should be correct layer if keyboard is toggled to `Win`

lith-x, 2024-03-04:
Sheesh, I feel dumb now, had no idea the `Win` setting was on layers 3, 4. Everything is working just fine. Thanks for the help, apologies!

closed 2024-03-04 by lith-x

exstntlmsnthrp, 2024-03-05:
> Sheesh, I feel dumb now, had no idea the `Win` setting was on layers 3, 4. Everything is working just fine. Thanks for the help, apologies!

Don't feel too bad. You're not alone. I did the exact same thing. It took me searching the internet for 3 days before I figured out I'd been changing layers 1 and 2. I swore something was wrong because my computer never recognized them. Then discovered I needed to change 3 and 4. I think it's something that could be more clearly documented by keychron.

PeterMortensen, 2024-12-28:
> on layers 3, 4

On the 3rd and 4th layer, but layer 2 and 3 in the normal designation (e.g., in Via), as the numbers start from 0.

For most keyboards, including the Q5 Pro:

  * Layer 2 and 3: Windows/[Linux](https://en.wikipedia.org/wiki/Linux)
    * Layer 2: Base layer
    * Layer 3: ***Fn*** key layer
  * Layer 0 and 1: Mac
    * Layer 0: Base layer
    * Layer 1: ***Fn*** key layer