Keychron/qmk_firmware #242: please sync wireless_playground with qmk/qmk_firmware/master

alex-vg, 2024-03-14:
### Issue Description

Hey! 👋 

First of all, thanks for developing open source! I got a Lemokey L3 and was missing the keyboard in the original project. The helpful Lemokey support pointed me to your fork.
I do see an issue here now. Why is the branch wireless_playground removing all other keyboards? (https://github.com/qmk/qmk_firmware/commit/4ae5990fcc0fabdbbb82e2e1c1375c861963c4d5) By doing so, you took away the possibility to merge all your keychron and lemokey keyboards back into the original qmk_firmware project. That branch is now also lacking behind and missing new features from qmk.

I would suggest to fix the removal in that commit (rewrite) then rebase onto master and then open a PR back to the original project. Or create a clean branch and copy the keyboard files if nothing else has changed. That way you can get all the updates from the people over at qmk and they profit by having your keyboards on the main branch.

scj643, 2024-04-02:
I want to eventually work on merging it with upstream I tried rebasing once and it failed miserably

PeterMortensen, 2024-04-26:
Related: [The current software state](https://github.com/Keychron/qmk_firmware/issues/217) (#217)

svenjacobs, 2024-05-30:
Also [related](https://github.com/Keychron/qmk_firmware/issues/145). This is a recurring question and so far we didn't get a satisfying answer by Keychron 😞

msbence, 2024-06-07:
I also tried merging upstream/master to the wireless_playground branch. It was pure suffering, but I managed to resolve all merge conflicts, and after removing some features (mainly the encoder) I could manage to build a firmware.
The reason I am making this comment is that after flashing this firmware I managed to brick my board. So merging upstream AND to make it work is even more complicated than I thought 😥 

Tip for whoever wants to try it: Eventhough my board was not going into bootloader mode with the Esc key method, unplugging the board (turning it off is not enough) and keeping the reset button pressed under the spacebar while inserting the cable did the trick, so this should unbrick your board.

PeterMortensen, 2024-06-22:
> keeping the reset button pressed under the spacebar while inserting the cable did the trick, so this should unbrick your board.

Yes, the space bar method is the backup method.

Though, allegedly, '[hard flashing](https://www.keychron.com/pages/how-to-flash-k8-pro-with-entire-chip-erase)' is sometimes necessary (essentially "`mass-erase:force`" added to the [dfu-util](https://man.archlinux.org/man/dfu-util.1.en) command line).

closed 2024-10-21 by github-actions[bot]