Keychron/qmk_firmware #244: [Bug] Flashing new firmware does nothing

OrangeLightning219, 2024-03-25:
### Describe the Bug

When flashing a new version of the firmware no chagnes are applied. For example I had [this keymap](https://github.com/OrangeLightning219/qmk_firmware/blob/4c3d562017dba77ac375731f700bcb8156fd5d9b/keyboards/keychron/q1_pro/ansi_knob/keymaps/OrangeLightning219/keymap.c) flashed then I made [this change](https://github.com/OrangeLightning219/qmk_firmware/commit/8a4723557befc957f66ee37e2e01c1166687360e) and flashed it using the qmk toolbox and nothing happend, the two keys didn't change. 

When I got this keyboard a few months ago everything was working fine. This started when I began experimenting with different layouts and adding more layers. Now it happens event if I go back to the default 4 layers.

The only thing I found that works is flashing the default firmware downloaded from [here](https://www.keychron.com/blogs/archived/how-to-factory-reset-or-flash-your-qmk-via-enabled-keychron-q1-pro-keyboard) and then flashing my keymap.

Any ideas what may be causing this behaviour?

### Keyboard Used

keychron_q1_pro_ansi_knob

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q1-pro-qmk-via-wireless-custom-mechanical-keyboard?variant=40441727385689

### Operating System

Windows

### qmk doctor Output

```text
qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: E:/Projects/qmk_firmware
Ψ Detected Windows 10 (10.0.22631).
Ψ QMK MSYS version: 1.7.2
Ψ Userspace enabled: False
Ψ Git branch: wireless_playground
Ψ Repo version: 0.23.1
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest wireless_playground: 2024-03-25 14:11:05 +0100 (8a4723557b) -- default qwerty
Ψ - Latest upstream/master: 2024-03-22 18:30:30 -0500 (4afbade6d1) -- Add ES_GRV to latam language-specific keycodes (#23333)
Ψ - Latest upstream/develop: 2024-03-24 00:20:05 +0000 (d0ee4a1cb2) -- Fix rgblight init (#23335)
Ψ - Common ancestor with upstream/master: 2023-12-18 13:53:02 +1100 (9539f135d8) -- Remove obvious user keymaps, `keyboards/[0-9]*` edition. (#22691)
Ψ - Common ancestor with upstream/develop: 2023-12-18 13:53:02 +1100 (9539f135d8) -- Remove obvious user keymaps, `keyboards/[0-9]*` edition. (#22691)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9ba)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2024-03-30:
what is `UNDERSCORE` and  `INVERSE_QUOTE` intend to do? I test the keymap with key tester, tapping `UNDERSCORE` registered `left shift` and `minus`, then unregistered  `minus`, keeped `left shift` being pressed.

OrangeLightning219, 2024-03-30:
`UNDERSCORE` is suposed to type `_` without pressing shift and `INVERSE_QUOTE` is suposed to invert the quote key - `"` without shift and `'` with shift. The shift beeing pressed after unregistering `minus` is a different issue I have (shift doesn't get unregistered even though there is [code ](https://github.com/OrangeLightning219/qmk_firmware/blob/4c3d562017dba77ac375731f700bcb8156fd5d9b/keyboards/keychron/q1_pro/ansi_knob/keymaps/OrangeLightning219/keymap.c#L98)to do that). The issue of flashing firmware not applying changes was happening even without these custom keycodes.

PeterMortensen, 2024-04-26:
This can be caused by the Via keymap overriding the QMK keymap. That is, if you save the Via configuration before the flashing and restore it afterwards.

Fix: Do the same change in the Via configuration (key mapping) as in the QMK key map. (Yes, this is a maintenance problem. Any change to the QMK key map proper must be applied in two places.)

OrangeLightning219, 2024-04-27:
Setting `VIA_ENABLE = no` in my `rules.mk` seems to have fixed the issue. I had it enabled because when I first got this keyborad I tried using the via app but it didn't even recognize the Q1 pro, so I switched to qmk. Thanks for your help.

closed 2024-04-27 by OrangeLightning219