Keychron/qmk_firmware #248: No VIA JSON provided for Keychron Q1 Pro

bjornorri, 2024-04-06:
### Describe the Bug

The [wireless_playground](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron) branch provides a `via_json` directory for virtually every Keychron keyboard for VIA compatibility.

However, no such directory is provided in the [`q1_pro` directory](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/q1_pro).

The VIA file provided in the [VIA pairing instructions](https://www.keychron.com/pages/how-to-use-via-to-pair-with-keychron-q-series-keyboard) does not work.

Can you provide these missing files for the Q1 Pro?

### Keyboard Used

keychron/q1_pro/ansi_knob

### Operating System

MacOS

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2024-04-12:
Added.

bjornorri, 2024-04-12:
Thank you for the quick response!

I was able to build the `wireless_playground` branch at commit e5e57f406e4281f3a9e61afdb6c3e7d323c0cf7b and use the [newly added JSON](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/q1_pro/via_json) to connect to VIA with no errors 👍

closed 2024-04-12 by bjornorri