Keychron/qmk_firmware #249: When wireless_playground branch will be merged to master?

RDSlaughter, 2024-04-07:
### Issue Description

**Hi everyone.**
I bought the v1 max model 2 weeks ago. Various errors are received when building QMK firmware with the current branch. I encountered other errors such as WS2812 pin definitions etc. and I fixed them one by one manually by removing the function definitions of some of them. I was able to get a successful find. 
I was able to define custom caps lock and layer rgb. But I could not use the charge indicator in any way. There are things that are out of date and constantly cause errors. I think we need to merge the **wireless_playground** branch to master.

alex-vg, 2024-04-10:
see https://github.com/Keychron/qmk_firmware/issues/242

lokher, 2024-04-12:
Did you build with keychron fork and `wireless_playground` branch? 
`wireless_playground` won't be merged to QMK upstream due to it make much core modification and they don't accept it.

RDSlaughter, 2024-04-12:
> Did you build with keychron fork and `wireless_playground` branch? `wireless_playground` won't be merged to QMK upstream due to it make much core modification and they don't accept it.

I did, i do not think it is ready to use.  At least for v1 max

closed 2024-08-11 by github-actions[bot]