Keychron/qmk_firmware #250: [Bug] Q10 Pro Bluetooth not working - Via keymap

GalaxyAllie, 2024-04-08:
### Describe the Bug

On my Q10 Pro, Bluetooth doesn't work when built from the `wireless_playground` branch. It connects fine, but keys give unexpected (and wrong) actions - like it's misinterpreting matrix scans.

### Keyboard Used

Keychron Q10 Pro

### Operating System

macOS

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [x] Karabiner (macOS)

### Additional Context

Works fine in USB, Karabiner isn't modifying events from this keyboard

lokher, 2024-04-12:
Can elabrate more and upload the compilred .bin file?

GalaxyAllie, 2024-04-18:
Hi @lokher - sorry for the delayed reply, I wasn't home, as I travel a lot I actually just bought a K11 Pro and have the same issue. What I mean is that when paired by Bluetooth (both to my MacBook Air and to my Pixel 7) most keys do nothing and those that do, don't do the right thing. For example `Q` sends `M` and `W` sends `A` on the K11 Pro.

I've tried a factory reset with no luck, and I've compiled from the head of the current branch to rule out any of the config options I've set. I can install the firmware from your website and it works fine. Unfortunately, there are some config options Via can't set I need (the whole reason I bought your keyboards - QMK support). 

I've attached a K11 Pro firmware I compiled from the HEAD of `wireless_playground` earlier today, which exhibits this issue. The start of the file name is the commit hash I compiled from.

[e5e57f406e4281f3a9e61afdb6c3e7d323c0cf7b_keychron_k11_pro_ansi_encoder_rgb_via.zip](https://github.com/Keychron/qmk_firmware/files/15030072/e5e57f406e4281f3a9e61afdb6c3e7d323c0cf7b_keychron_k11_pro_ansi_encoder_rgb_via.zip)

PeterMortensen, 2024-04-26:
This *may* be caused by the keyboard [being in (full) NKRO mode](https://www.reddit.com/r/Keychron/comments/198jmqn/comment/l0vwnbr/).

Recover by ***Fn*** + ***N*** (if using the default key mapping). If the firmware has been changed to use "#define FORCE_NKRO" then it must be done at every power up.

GalaxyAllie, 2024-04-28:
Thanks @PeterMortensen I'm using `#FORCE_NKRO`, yes - now not when I'm on a freshly compiled version but NKRO would have remained and I didn't even think about it.

GalaxyAllie, 2024-04-29:
Hi, I can confirm that this is the issue and disabling NKRO allows Bluetooth to work correctly - obviously not ideal so I'll leave this open and update the issue title

nskobelevs, 2025-06-25:
I ran into this a few days ago and ended up looking into it.

It looks like the `send_nkro` implementation, [`bluetooth_send_nkro`](https://github.com/Keychron/qmk_firmware/blob/wireless_playground/keyboards/keychron/bluetooth/bluetooth.c#L326-L329) for the bluetooth host driver is using `send_keyboard` instead of `send_nkro` on the Bluetooth transport so it ends up sending a `CKBT51_CMD_SEND_KB` command to the bluetooth chip but with the nkro bits structure payload instead of keyboard report payload causing the misinterpretation. 

>For example Q sends M and W sends A on the K11 Pro.

`KC_Q` = `0x14` = `(2 << 3) | 4` which for nkro would set the 4th bit in the 2nd byte (i.e. `0b00010000` = `0x16`) so we get a `KC_M` instead.

Similarly,
`KC_W` = `0x1a` = `(3 << 3) | 2` which for nkro would set the 2nd bit in the 3rd byte (i.e. `0b00000100` = `0x04`) so we get a `KC_A` instead.

This isn't an issue on the old `bluetooth_playground` branch as back then nkro didn't have it's own report so this was likely introduced when splitting `bluetooth_send_nkro` from `bluetooth_send_keyboard`.

Fix PR: #394