Keychron/qmk_firmware #255: [Bug] Enabling NKRO on K3 Max causes random duplicated key presses in 2.4GHz mode

hlysine, 2024-04-22:
### Describe the Bug

Enabling NKRO via <kbd>Fn</kbd>+<kbd>N</kbd> on the K3 Max causes keys presses to be randomly duplicated even though no typos are made. The amount of duplications seems to correlate with typing speed.

This does not happen if the keyboard is connected via cable or if NKRO is disabled. The keyboard still correctly detects 20+ simultaneous key presses in 2.4GHz+NKRO despite this bug.

Example text:
```
The quick brown fox jumps over the lazy dog.
```
With 2.4GHz+NKRO (multiple tries at ~100WPM):
```
The  quick b rown fox jumps over the lazy dog.
The q uicckc brown fox jumps over  the lazyz dodg.
The  quickc brown fox jumps o ver the lazy dogg.
The quick browon fox j umps o ver the lazy dog.
The  quick broowon fox jumps over the lazy dog.
```
With 2.4GHz+NKRO (multiple tries at ~30WPM):
```
The quick brown fox jumps over the lazy dog.
The q uick brown fox jumps over the laz ydogg.
The quick brown fox jumps over the lazy dog.
The quick brown fox jumps over the laz ydog.
```

### Keyboard Used

`keychron/k3_max/ansni/rgb/keymaps/via`

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k3-max-qmk-via-wireless-custom-mechanical-keyboard

### Operating System

Windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: D:/Projects/QMK/qmk_firmware
Ψ Detected Windows 10 (10.0.22631).
Ψ QMK MSYS version: 1.8.0
Ψ Userspace enabled: False
Ψ Git branch: wireless_playground
Ψ Repo version: 0.14.29
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9ba)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

None

PeterMortensen, 2024-05-31:
Possibly related: *[\[Bug\] Q10 Pro and K11 Pro Bluetooth not working with NKRO - Via keymap #250](https://github.com/Keychron/qmk_firmware/issues/250)*

OloRatuj, 2024-09-23:
I am suffering the same issue with K15 Max

failsafe42, 2024-10-05:
Just got my K17 Max today and I noticed the same issue. 6KRO wireless works fine and so does NKRO wired, but NKRO wireless seems to cause double presses if you type too quickly.

bsmely, 2025-03-16:
+1
K17 Max, the same - NKRO wireless cause double presses

K17 Max ANSI Knob RGB v1.0.0
LKBT v0.2.1