Keychron/qmk_firmware #256: [Feature Request] Q1 Max: german DE Layout

Andalador, 2024-04-29:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi, unfortunately there is no de (german) layout for the Q1 Max. Is someone able to change this?

Any help would be much appreciated. Unfortunately I don't know enough about programming to do so myself.

adophoxia, 2024-04-30:
> there is no de (german) layout for the Q1 Max. Is someone able to change this?

What do you mean by this exactly?

Andalador, 2024-04-30:
> What do you mean by this exactly?

What I mean is that this can be used as an iso layout for the Q1 max, but the "right part" of the german layout is nowhere to be found.

Here is an example: [here at the top](https://www.mechanischetastatur.guide/wp-content/uploads/2022/02/iso_VS_ansi-1024x1024.png) 

I'm talking specifically about the ü, ö, ä letters. I can use the keyboard as is, but I can't put those letters in any sort of macro or anything because they simply don't show up anywhere in via.

Hope this helped?

PeterMortensen, 2024-07-08:
Here is [a good explanation](https://www.reddit.com/r/ErgoMechKeyboards/comments/1dmmrwj/comment/l9wqsn8/).

For example, if you use KC_SCLN (or ";" if it is in a string) in a macro and the keyboard layout in the operating system is set to German, the result will be "ö".

Yes, it is confusing that an US keyboard layout is assumed, e.g., in Via (Vial has an interpretation for a German physical key layout (menu *Keyboard layout* &rarr; *German QWERTZ*. And even a Swiss German one), though it doesn't change the actual behaviour of the keyboard, only how it is presented in Vial, and thus makes it less confusing.) But nothing needs to change in the firmware to achieve what you want to achieve.

Note that there are two extra key codes on the ISO layout, KC_NUBS (the key between Shift and "Z"/"Y") and KC_NUHS. See the key map.

closed 2024-11-07 by github-actions[bot]