Keychron/qmk_firmware #259: [Bug] Latest playground is broken on Keychron Q6 (at least). Keys messed up/swapped.

richard-browne, 2024-05-06:
### Describe the Bug

I built the latest on playground with `make keychron/q6/ansi_encoder:keychron`. After installing the firmware many keys were messed up. Control and fn are swapped, all keys on the asdf row are off by one (e.g. a key gives 's', s key gives 'd' etc).

I went back to an earlier commit and that worked fine.

### Keyboard Used

Q6

### Operating System

n/a

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2024-05-08:
Is your Q6 with knob or without knob? The knob version use `make keychron/q6/ansi_encoder:keychron`, or else use `make keychron/q6/ansi:keychron`

richard-browne, 2024-05-08:
Yes I have the knob version. I built the latest on old_playground and that worked fine.

I wish Keychron would tag the commits corresponding to the official firmware releases.

richard-browne, 2024-06-01:
Keychron should do better. There's no way to know what branch/commit to build for your keyboard. It's a mess.

PeterMortensen, 2024-06-22:
Is there a reason not to use [the official QMK repository](https://github.com/qmk/qmk_firmware/tree/master/keyboards/keychron/q6)? The raw hid commands? Support by [Keychron's Via clone](https://www.launcher.keychron.com/#/keymap)? Something else?

It was my impression it was kept most up-to-date there.