Keychron/qmk_firmware #262: [Bug] After wake from sleep MOD_TAP key codes require double tap

slasi, 2024-05-17:
### Describe the Bug

My keyboard is connected to my Mac via USB cable. I have my caps lock key programmed using the "Any Key" function to MT(MOD_LCTL, KC_ESC). This works great: single tap for escape and hold for ctrl. After my Mac goes to sleep and I wake it up it no longer works properly. After sleep, single tap does not issue escape key. I have to double tap for escape key. This is 100% reproducible on my system

### Keyboard Used

keychron/k3_max/ansi/rgb

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k3-max-qmk-via-wireless-custom-mechanical-keyboard?variant=40685197033561

### Operating System

macOS 14.4.1

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

slasi, 2024-05-22:
I was sent a new firmware file from keychron support which fixes the issue. Is this fix available in this repository? I'd like to make some custom modifications. Thanks.

PeterMortensen, 2025-03-24:
That was one day after [this change](https://www.reddit.com/r/olkb/comments/1dyetan/comment/lccz0dv/). Coincidence?