Keychron/qmk_firmware #264: K11 Max Firmware

boobay, 2024-05-23:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

When will the k11 max code be available under wireless playground? :)

jnolck, 2024-06-02:
Me two. I just got the k11 max like two weeks ago and had good scare the night I got it. I compiled and flashed the k11 pro firmware onto it and thought I bricked it.

boobay, 2024-06-02:
Customer support says the source code will be available within a few weeks in the wireless_playground branch.

choenig, 2024-07-03:
It looks like it landed :)

https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/k11_max

closed 2024-11-01 by github-actions[bot]