Keychron/qmk_firmware #266: Lemokey P1 source code missing

madmalkav, 2024-05-26:
### Issue Description

I'm interested in purchasing this keyboard but I can't find the source code in the Lemokey folder of this repository.

lokher, 2024-08-20:
Added.
https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/lemokey/p1

GreyXor, 2024-09-09:
Hello @lokher,
Why don't you work directly on github? I mean, you made version 1.0.1 and 1.0.2. But this github repo hasn't been updated. The idea of open source is not just to put the source code online, but also to have open governance.
Please can Keychron can :
- Update this repo to latest version 1.0.2
- Add ISO variant of P1
- Keep updating the repo
Thanks!

GreyXor, 2025-01-02:
@lokher hi, can we have the latest ISO firmware source code ?

lokher, 2025-01-04:
> @lokher hi, can we have the latest ISO firmware source code ?

I will add it later. Most keychron ANSI source can be used to add ISO, the differece is there is a between left shift and Z, so the steps are as below:
### 1. Copy the "ansi_encoder" to "iso_encoder"
### 2. Edit "iso_encoder/info.json" to add a key to layout, 

- change below lines 

```                
                {"matrix": [4, 0], "x": 0, "y": 4.25, "w": 2.25},
                {"matrix": [4, 2], "x": 2.25, "y": 4.25},
```
to 
```               
                {"matrix": [4, 0], "x": 0, "y": 4.25, "w": 1.25},
                {"matrix": [4, 1], "x": 1.25, "y": 4.25},
                {"matrix": [4, 2], "x": 2.25, "y": 4.25},
```

- change layout name `LAYOUT_ansi_82` to `LAYOUT_iso_83`

### 3. Add keycode to `keymap.c`
change ` KC_LSFT,            KC_Z, ` to ` KC_LSFT,  KC_NUBS,  KC_Z, ` 
and swap enter key and back slash
change 'KC_ENT` to `KC_NUHS`, change `KC_BSLS` to `KC_ENT`
### 4 change `#    define RGB_MATRIX_LED_COUNT 81` to `#    define RGB_MATRIX_LED_COUNT 82` in `config.h`
### 5. Rename `ansi_encoder.c` to `iso_encoder.c` and edit

- Add line   ` {1, G_14,   I_14,   H_14},` between 

```  
    {1, G_15,   I_15,   H_15},
    {1, G_13,   I_13,   H_13},
```

- change 

```        
        { 58, __, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, __},
        { 71, 72, 73, __, __, __, 74, __, __, 75, 76, 77, 78, 79, 80 },
```
to 
```       
        { 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, __},
        { 72, 73, 74, __, __, __, 75, __, __, 76, 77, 78, 79, 80, 81},
```

- change 

` {8,49},          {33,49}, {48,49}, {62,49}, {77,49}, {92,49}, {106,49}, {121,49}, {136,49}, {150,49}, {165,49}, {186,49}, {203,52},`
to 
` {5,49}, {20,49}, {33,49}, {48,49}, {62,49}, {77,49}, {92,49}, {106,49}, {121,49}, {136,49}, {150,49}, {165,49}, {186,49}, {203,52},`

change 
`        1,    1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,`
to 
`        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,`

srouquette, 2025-03-10:
is it compatible with Lemokey P1 HE ?

L0nelyE4rth, 2025-03-11:
> is it compatible with Lemokey P1 HE ?

No, and that HE device live in Hall_effect_playground

closed 2025-07-10 by github-actions[bot]