Keychron/qmk_firmware #272: [Bug] Unable to build K3 Max because of 'function with qualified void return type called'

bbucko, 2024-06-11:
### Describe the Bug

I think that commit b6d42d84d2726e3976508dac48f937b2502f03e8 broke the build for K3 Max. 

### Keyboard Used

k3_max/ansi

### Operating System

mac

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

bbucko, 2024-06-24:
Broken toolchain on my Mac.

closed 2024-06-24 by bbucko

bilogic, 2025-01-02:
@bbucko 

can i ask 

1. if you used to wireless_playground branch?
2. were the build steps as per QMK's default or something different?

thanks!

bbucko, 2025-01-02:
Issue was caused by a different gcc installed on my Mac. I reinstalled the packages and the issue went away.

bilogic, 2025-01-03:
cool thanks let me try to build and flash my K3 Max