Keychron/qmk_firmware #273: Publish sources for Q3 Max ISO

inkfarer, 2024-06-18:
I'd like to request the source files for the [ISO version](https://www.keychron.com/collections/keychron-q-max-series-keyboard/products/keychron-q3-max-qmk-via-wireless-custom-mechanical-keyboard-iso-layout-collection) of the Q3 Max. I've found that only the ANSI version has sources available as of now.

DottieUnderwood, 2024-06-24:
Same for Q6 Max ISO. The keyboard is shipping but the source is nowhere to be found :-(

mvbentes, 2024-07-25:
Same for K13 Pro ISO. I need to be able to set `TAPPING_TERM` for mod tap and that cannot be accomplished via the launcher.
Both the `bluetooth_playground` and the `wireless_playground` branches only contain the ANSI source.

lokher, 2024-08-19:
> Same for Q6 Max ISO. The keyboard is shipping but the source is nowhere to be found :-(
https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/q6_max/iso_encoder

closed 2024-12-19 by github-actions[bot]