Keychron/qmk_firmware #274: Please add the Keychron Q1 HE source code

Stopmodrifts, 2024-06-23:
### Issue Description

I am really trying to customize this keyboard with QMK but I cannot do it without the source code.

CuteSC2, 2024-06-23:
Yea it would be awesome too get the Q1 he firmware soon, want to make a custom one with per key repeat rate and delay.

PeterMortensen, 2024-06-30:
It could potentially [be a six-month wait](https://www.reddit.com/r/Keychron/comments/1dj1180/comment/l9ffcl0/).

lokher, 2024-08-19:
Added.
https://github.com/Keychron/qmk_firmware/tree/hall_effect_playground/keyboards/keychron/q1_he

PeterMortensen, 2024-08-22:
Related:  *[#268](https://github.com/Keychron/qmk_firmware/issues/268)*

Shahaed, 2024-10-07:
Added here: https://github.com/Keychron/qmk_firmware/tree/hall_effect_playground/keyboards/keychron/q1_he

Please close this issue

closed 2025-02-06 by github-actions[bot]