Keychron/qmk_firmware #278: [Bug] only bluetooth mode k1 max

paperhand, 2024-07-04:
### Describe the Bug

wireless_playground/keyboard/keychron/k1_max/ansi/rgb 
cant change mode. fixed in blutooth mode.
I tried connect C cable. but it doesnt work. 2.4 also doesnt work.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

PeterMortensen, 2024-07-08:
Change mode by what means? The (physical) switch on the left of the keyboard? Or something else? 

How is it related to the firmware?

Can you elaborate?

ievsdoma, 2024-07-24:
I think I am having a similar issue with Q12 Max: https://github.com/Keychron/qmk_firmware/issues/288. Hope that provides additional information.