Keychron/qmk_firmware #280: [Bug][K3 Pro] Issue with Right arrow partially working

AurelienKun, 2024-07-12:
### Describe the Bug

I'm using a K3 Pro ISO RGB (FRA layout) and the right arrow key does not respond unless doing compo key (like shift+right) or using another arrow first within few ms.
I'm using stock firmware.

### Link to product page (if applicable)

https://keychronhk.com/products/keychron-k3-pro-qmk-via-wireless-custom-mechanical-keyboard?variant=44665338462527

### Operating System

Fedora 40

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2024-07-15:
Sound like there is bug, plz try this firmware 
[keychron_k3_pro_iso_rgb_via.zip](https://github.com/user-attachments/files/16229850/keychron_k3_pro_iso_rgb_via.zip)

jsuarezarm, 2024-08-06:
I have the same problem. In my case, it is through the Bluetooth connection. More specifically, it happens with the keys on the right column of the keyboard (backlight, page up, page down, home, end, right arrow).

When the backlighting is enabled, these keys function until the keyboard is automatically turned off (auto-sleep function). When the backlighting is disabled, these keys do not wake up the keyboard after 1 second without pressing another key.