Keychron/qmk_firmware #286: [Feature Request] More or less reproducible builds for supplied firmware images

BlackLotus, 2024-07-22:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

I'm using the k8 pro as an example here.
When the official firmware file for the k8 pro (k8_pro_us_rgb_v1.03.bin) is flashed via works like expected. When building the k8 pro firmware with via support no matter which branch I use I can't reproduce this firmware.
I'm not even talking about having the same hash, I'm talking about working like intended. I can't configure my keyboard via via when building from source even when supplying the appropriate json file.

It would be nice if in the firmware description there would be a pin to a commit hash or a tar archive of a build environment that can reproducible produce a working image equal to the provided binary files.

PeterMortensen, 2024-08-29:
In what way does Via not work as expected when self-building the firmware?

PeterMortensen, 2024-11-15:
Though the holding down ***Fn*** + ***J*** + ***Z*** for 3 seconds to reset to factory defaults does ***not*** work with self-compiled firmware (by default, anyway).

closed 2025-03-16 by github-actions[bot]